stlink-org/stlink #465: Building on Windows with MinGW64

ivankravets, 2016-09-15:
Hi,

According this PR https://github.com/texane/stlink/pull/434 the MinGW should work. However, it doesn't work for me :(

Could someone help? Thanks!

* CMakeError.log http://pastebin.com/9m6U7yZD
* CMakeOutput.log http://pastebin.com/SqvEJ41a

I've tried to use `mman` from this repo https://github.com/witwall/mman-win32 but it didn't help :(

xor-gate, 2016-09-15:
I will enable appveyor builds for my fork. As @texane hasn't enabled this (yet). Probably I have broken something with this test: https://github.com/texane/stlink/commit/57cf59a86930b37d0d408fc58b6f13b44c8a431a

ivankravets, 2016-09-15:
Thanks, will wait for the updates.

xor-gate, 2016-09-15:
Which version are you building?

xor-gate, 2016-09-15:
It seems broken as you mention: https://ci.appveyor.com/project/xor-gate/stlink/build/job/eehlrjinv5codm44

xor-gate, 2016-09-15:
It seems almost to work now, there was a bug in the cmake file.
https://ci.appveyor.com/project/xor-gate/stlink/build/4/job/farxcgt4j3mbwajq

```
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_poll':
c:/projects/stlink/src/mingw/mingw.c:50: undefined reference to `_imp__select@20'
c:/projects/stlink/src/mingw/mingw.c:61: undefined reference to `__WSAFDIsSet@8'
c:/projects/stlink/src/mingw/mingw.c:63: undefined reference to `__WSAFDIsSet@8'
c:/projects/stlink/src/mingw/mingw.c:65: undefined reference to `__WSAFDIsSet@8'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_socket':
c:/projects/stlink/src/mingw/mingw.c:113: undefined reference to `_imp__socket@12'
c:/projects/stlink/src/mingw/mingw.c:115: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_connect':
c:/projects/stlink/src/mingw/mingw.c:127: undefined reference to `_imp__connect@12'
c:/projects/stlink/src/mingw/mingw.c:130: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_accept':
c:/projects/stlink/src/mingw/mingw.c:143: undefined reference to `_imp__accept@12'
c:/projects/stlink/src/mingw/mingw.c:145: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_shutdown':
c:/projects/stlink/src/mingw/mingw.c:159: undefined reference to `_imp__shutdown@8'
c:/projects/stlink/src/mingw/mingw.c:162: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_close_socket':
c:/projects/stlink/src/mingw/mingw.c:169: undefined reference to `_imp__closesocket@4'
c:/projects/stlink/src/mingw/mingw.c:171: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_write_socket':
c:/projects/stlink/src/mingw/mingw.c:178: undefined reference to `_imp__send@16'
c:/projects/stlink/src/mingw/mingw.c:180: undefined reference to `_imp__WSAGetLastError@0'
CMakeFiles/stlink.dir/objects.a(mingw.c.obj): In function `win32_read_socket':
c:/projects/stlink/src/mingw/mingw.c:187: undefined reference to `_imp__recv@16'
c:/projects/stlink/src/mingw/mingw.c:189: undefined reference to `_imp__WSAGetLastError@0'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [libstlink.dll] Error 1
make[1]: *** [CMakeFiles/stlink.dir/all] Error 2
make: *** [all] Error 2
Command exited with code 2
```

xor-gate, 2016-09-15:
Wohoo it seems to work:

```
C:\Users\Jerry\Downloads\stlink\build (master)
λ st-info.exe --probe
Found 1 stlink programmers
 serial: 503f6b06506752554231033f
openocd: "\x50\x3f\x6b\x06\x50\x67\x52\x55\x42\x31\x03\x3f"
  flash: 262144 (pagesize: 256)
   sram: 32768
 chipid: 0x0427
  descr: L1 Medium-Plus-density device
```

xor-gate, 2016-09-15:
Here is the build output: https://ci.appveyor.com/project/xor-gate/stlink/build/10.

ivankravets, 2016-09-15:
1. My Win7 in VM is clean
2. I use MinGW64 the same as you specified in [here](https://github.com/texane/stlink/blob/master/scripts/appveyor-mingw.sh)
3. `mkdir build && cd build`
4. `cmake .. -G"MinGW Makefiles"`

See logs:
* http://pastebin.com/mdTs9i5g
* http://pastebin.com/GfjF4GuT

What I'm doing wrong? Thanks!

xor-gate, 2016-09-15:
I'm not sure, the output you have pasted has nothing todo with your problem. CMake tried to check compiler options by compiling a dummy file with those options. This is why you see some fails. You should provide the output of the `cmake -G` run.

https://github.com/texane/stlink/blob/master/scripts/mingw64-build.bat

ivankravets, 2016-09-15:
1. Removed totally all files with which I've played today
2. Cloned repo from the scratch.
3. Build and Success!!!

Thank you!

closed 2016-09-15 by ivankravets

xor-gate, 2016-09-15:
I am so happy you have it working! Thanks for your patience and reporting.