stlink-org/stlink #471: gdb-server.c: cannot recv: -2

424778940z, 2016-09-17:
System: Archlinux latested
IDE: Qt Creator
Compiler: arm-none-eabi-gcc 6.2.0
Debugger: arm-none-eabi-gdb 7.11.1

Steps:
1. compile project 
2. click Run
3. st-link works perfectly
4. modify code
5. compile project (success)
6. click Run again
7. error happend....

`2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/common.c: *** stlink_read_mem32 ***
2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: send: bd46
2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: recv: qL1200000000000000000
2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: send: 
2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: recv: vKill;a410
2016-09-17T17:04:29 DEBUG /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: send: OK
2016-09-17T17:04:29 ERROR /home/adam/Desktop/stlink-master/src/gdbserver/gdb-server.c: cannot recv: -2
2016-09-17T17:04:30 DEBUG /home/adam/Desktop/stlink-master/src/common.c: *** stlink_run ***
2016-09-17T17:04:30 DEBUG /home/adam/Desktop/stlink-master/src/common.c: *** stlink_exit_debug_mode ***
2016-09-17T17:04:30 DEBUG /home/adam/Desktop/stlink-master/src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2016-09-17T17:04:30 DEBUG /home/adam/Desktop/stlink-master/src/common.c: *** stlink_close ***
`

xor-gate, 2016-09-18:
Probably the target is still running code and the core is not halted. Then the programmer USB is unable receive/send any command over SWD.

424778940z, 2016-09-22:
Is there a way only download the firmware and reset, but not debug?

xor-gate, 2016-09-22:
You could use `st-flash` tool.

xor-gate, 2017-01-14:
If there are any questions left, feel free to post them here. Probably you have figured out how to flash your board in the meantime as this issue is a few months old now. Thanks you for your question.

closed 2017-01-14 by xor-gate