stlink-org/stlink #472: Problems with debian package generation

OsterlaD, 2016-09-18:
- [X] Operating system: Linux
- [X] git commit hash: 1.2.0-118-g7023a9a

* debuild failed with "can not find install files" message
* man pages not installed

xor-gate, 2016-09-18:
Could you try the `cmake-fixes` branch? Probably this needs some work because I'm developing currently from a mac. Probably I have broken the debian packaging. Could you give more details how you package stlink?

OsterlaD, 2016-09-18:
I will have a look at this branch.
Can you have a look on pull request 473.
These are my changes to fix broken install and some warnings.

xor-gate, 2016-09-18:
Looks good, I only see no reason the have a duplicate copyright file into the debian folder. Is there no way to use the LICENSE file and copy it over in generation to debian folder? As mentioned in the pull request you should rely only for man pages on `doc/man/*.1` and not the `build/Debug` folder. If you fix this I will merge.

OsterlaD, 2016-09-18:
A good point,
two possible solutions:
* a symbolic link, maybe not the best solution, but simple
* copy it at build time, I have to figure out when and how

No problem, I will remove build/Debug prefix in manpages file.

Shall I rebase patch stack on cmake-fixes?

xor-gate, 2016-09-18:
You may rebase on cmake-fixes but it will also apply on `master` because you only changed stuff in the `debian` folder. The copyright file should be formatted as follows (and not include the LICENSE file from the project)

https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/

OsterlaD, 2016-09-18:
Is doc/man/st-util.1 from "fabien.lementec I fix: add missing st-util.1 until proper solution" really the better than they from build/Debug/doc/man/st-*.1?

OsterlaD, 2016-09-18:
```
W: stlink-tools: manpage-has-bad-whatis-entry usr/share/man/man1/st-util.1.gz
W: stlink-tools: manpage-has-errors-from-man usr/share/man/man1/st-util.1.gz 1: warning: macro `PP' not defined
W: stlink-tools: binary-without-manpage usr/bin/st-flash
W: stlink-tools: binary-without-manpage usr/bin/st-info
```

OsterlaD, 2016-09-18:
You are right about copyright.
Thanks for this hint.

OsterlaD, 2016-09-18:
Reworked patch stack on master.
Now manpages contains doc/man/st-*.1 entry, corresponding to install entry usr/bin/st-*.
The debian/copyright file indicates BSD-3-clause license with copyright holders from LICENSE file.

OsterlaD, 2016-09-18:
Encountered a problem with rename of lib.

OsterlaD, 2016-09-18:
Removed this patch from pull request.
Seems it have affects on cmake build.
It requires more then just rename.
This is not very important.

xor-gate, 2016-09-19:
Thank you for your contribution!

Nightwalker-87, 2020-03-17:
Finally closed by #485.

closed 2020-03-17 by Nightwalker-87