stlink-org/stlink #475: ST-LINK RST pin

SL-RU, 2016-09-18:
Hello!
I'm now required in force hardware reset of stm32 before debug using optional RST pin on the st-link. In the keil this option just works by default, but st-utility don't use RST at all. Is that feature implemented? If not, is is possible to implement?

Thank you,
The stlink project maintainers

xor-gate, 2016-09-18:
You should use `--reset` this triggers a RST pin reset IIRC.

SL-RU, 2016-09-18:
```
~$ sudo st-util --reset
st-util 1.2.0-119-g4209e4b ((null))
st-util: unrecognized option '--reset'
2016-09-18T23:31:36 INFO /home/lyra/prg/stlink/src/common.c: Loading device parameters....
2016-09-18T23:31:36 WARN /home/lyra/prg/stlink/src/common.c: unknown chip id! 0
```
Didn't work =c

xor-gate, 2016-09-21:
With st-util it should automaticly reset the target when started:
https://github.com/texane/stlink/blob/master/src/gdbserver/gdb-server.c#L181
https://github.com/texane/stlink/blob/master/src/gdbserver/gdb-server.c#L202-L204\\\

Could you try adding:
https://github.com/texane/stlink/blob/6dadc9faa1e899b6f7a37c92f1c7fe4f74b7dd75/src/tools/flash.c#L189

And see if this helps?

SL-RU, 2016-09-24:
Yep, that helped! Maybe it need to be included in the release by default or by arguments?

xor-gate, 2016-09-25:
I will add this behaviour in v.1.3.0 then it is the same as `st-flash`. Only this is available for stlink/v2 programmers.

xor-gate, 2016-12-30:
I have just merged a possible fix from PR #534. Could you test latest master?

SL-RU, 2016-12-30:
Yep, now it's working fine, thanks!

xor-gate, 2016-12-30:
Thanks, I will close this then!

closed 2020-03-17 by Nightwalker-87