stlink-org/stlink #477: unknown chip id! 0 again...

SL-RU, 2016-09-19:
- Stlink/v2
- where can I find version of st-link?
- Debian Jessie
- st-util 1.2.0-119-g4209e4b ((null))
- st-util
- STM32F103C8T6, STM32F100C4T6

1) STM32 has no program in flash. Then I'm connecting it to ST-Link, then opening st-util and flash using 'load' in gdb. Everything OK.
2) After flashing I'm disconnecting gdb, then st-link from board and powering it off.
3) After doing step 1 again output of st-util looks like this:
```
~$ sudo st-util 
st-util 1.2.0-119-g4209e4b ((null))
2016-09-19T20:39:07 INFO /home/lyra/prg/stlink/src/common.c: Loading device parameters....
2016-09-19T20:39:07 WARN /home/lyra/prg/stlink/src/common.c: unknown chip id! 0
```
4) If I'm holding reset button & executing st-util:
```
~$ sudo st-util 
st-util 1.2.0-119-g4209e4b ((null))
2016-09-19T20:51:54 INFO /home/lyra/prg/stlink/src/common.c: Loading device parameters....
2016-09-19T20:51:54 INFO /home/lyra/prg/stlink/src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2016-09-19T20:51:54 INFO /home/lyra/prg/stlink/src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0 bytes (0 KiB) in pages of 1024 bytes
2016-09-19T20:51:54 INFO /home/lyra/prg/stlink/src/gdbserver/gdb-server.c: Chip ID is 00000410, Core ID is  1ba01477.
2016-09-19T20:51:54 INFO /home/lyra/prg/stlink/src/gdbserver/gdb-server.c: Listening at *:4242...
```
Everything seems okay, but it doesn't:
```
(gdb) load
Loading section .isr_vector, size 0x10c lma 0x8000000
Load failed
(gdb)
```
I can erase flash using st-flash, but only if I'm holding reset. After erasing everything it starts again from 1 step.

Thank you,
The stlink project maintainers

SL-RU, 2016-09-19:
Ooooh, I see. Sorry for duplicated issue....

Nightwalker-87, 2020-03-17:
This is not a duplicated issue.

Nightwalker-87, 2020-03-17:
This is not a duplicate, but obviously related to #517 which seems to resolve the issue.

closed 2020-03-17 by Nightwalker-87