stlink-org/stlink #478: Broken dependencies in installed files

duro80, 2016-09-20:
- [ ] Operating system: Linux, Ubuntu 16.04
- [ ] Stlink tools version:  from HASH b5bbf3d
- [ ] Stlink commandline tool name: all programms

The problem is from commit b5bbf3d until last commit. 
I build st-link programs from source. Everythink is OK. After installing "sudo make install", I ran the program "st-info" (or other st-xxxx program) and this is output:

```
st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory.

```
I try the installation procedure on 3 different PC, same result.

xor-gate, 2016-09-20:
Could you run `ldd /usr/local/bin/st-info`  ? probably the references to the shared library are not good. I see `st-info` is looking for `libstlink.so.1` is it installed under `/usr/local/lib`? Maybe `ld-config` needs to be run before it is found.

xor-gate, 2016-09-20:
It works for me:

 make install to custom destination dir:

```
jjacobs@dev03:~/repos/github/stlink/build/Release$ make install DESTDIR=_install
[ 20%] Built target stlink
[ 41%] Built target stlink-static
[ 50%] Built target st-flash
[ 55%] Built target st-info
[ 67%] Built target st-util
[ 73%] Built target stlink-gui
[ 79%] Built target stlink-gui-local
[ 85%] Built target usb
[ 91%] Built target sg
[100%] Built target flash
Install the project...
-- Install configuration: "Release"
-- Up-to-date: _install/usr/local/lib/libstlink.so.1.2.0
-- Up-to-date: _install/usr/local/lib/libstlink.so.1
-- Up-to-date: _install/usr/local/lib/libstlink.so
-- Up-to-date: _install/usr/local/lib/libstlink.a
-- Up-to-date: _install/usr/local/bin/st-flash
-- Up-to-date: _install/usr/local/bin/st-info
-- Up-to-date: _install/usr/local/bin/st-util
-- Up-to-date: _install/usr/local/bin/stlink-gui
-- Up-to-date: _install/usr/local/share/stlink/stlink-gui.ui
-- Up-to-date: _install/usr/local/lib/pkgconfig/stlink.pc
-- Up-to-date: _install/usr/local/include/stlink.h
-- Up-to-date: _install/usr/local/include/stlink/backend.h
-- Up-to-date: _install/usr/local/include/stlink/chipid.h
-- Up-to-date: _install/usr/local/include/stlink/commands.h
-- Up-to-date: _install/usr/local/include/stlink/flash_loader.h
-- Up-to-date: _install/usr/local/include/stlink/logging.h
-- Up-to-date: _install/usr/local/include/stlink/mmap.h
-- Up-to-date: _install/usr/local/include/stlink/reg.h
-- Up-to-date: _install/usr/local/include/stlink/sg.h
-- Up-to-date: _install/usr/local/include/stlink/usb.h
-- Up-to-date: _install/usr/local/include/stlink/version.h
-- Up-to-date: _install/usr/local/share/man/man1/st-util.1
-- Up-to-date: _install/usr/local/share/man/man1/st-flash.1
-- Up-to-date: _install/usr/local/share/man/man1/st-info.1
-- Up-to-date: _install/usr/local/share/man/man1/st-term.1
```

dynamic linker (ldd) is unable to find the library:

```
jjacobs@dev03:~/repos/github/stlink/build/Release$ ldd _install/usr/local/bin/st-info
        linux-vdso.so.1 (0x00007fff4e7b9000)
        libstlink.so.1 => not found
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f0db46b3000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0db4308000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f0db40f9000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0db3edc000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f0db48cb000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0db3cd4000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0db3ad0000)
```

exporting a custom library path:

```
jjacobs@dev03:~/repos/github/stlink/build/Release$ export LD_LIBRARY_PATH=$PWD/_install/usr/local/lib
jjacobs@dev03:~/repos/github/stlink/build/Release$ ldd _install/usr/local/bin/st-info
        linux-vdso.so.1 (0x00007fff5bdc3000)
        libstlink.so.1 => /home/jjacobs/repos/github/stlink/build/Release/_install/usr/local/lib/libstlink.so.1 (0x00007f1dfe4e3000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f1dfe2cb000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1dfdf20000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f1dfdd11000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1dfdaf4000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f1dfe6f6000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1dfd8ec000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1dfd6e8000)
```

running `st-info --version`

```
jjacobs@dev03:~/repos/github/stlink/build/Release$ ./_install/usr/local/bin/st-info --version
v1.2.0-125-g7b358fe
```

duro80, 2016-09-20:
Output from ldd /usr/local/st-info:

>  linux-vdso.so.1 => (0x00007ffd4e7f0000)
>  libstlink.so.1 => not found
>  ...

Solution was:
> export LD_LIBRARY_PATH=/usr/local/lib

but I think, this is hack. It can be in installation script or in linker settings.

xor-gate, 2016-09-20:
This is not a hack, because package maintainers always run `ld-config` after a library is installed. Then the library is added to the cache `/etc/ld.so.cache`. You must make sure `/usr/local/lib` is added to the `ld.so.conf` search paths. On debian 8 it is configured here:

```
jjacobs@dev03:~$ cat /etc/ld.so.conf
include /etc/ld.so.conf.d/*.conf

jjacobs@dev03:~$ cat /etc/ld.so.conf.d/libc.conf
# libc default configuration
/usr/local/lib
```

You should read the man-page of [`ld-config`](https://linux.die.net/man/8/ldconfig) and [`ldd`](http://man7.org/linux/man-pages/man1/ldd.1.html)

xor-gate, 2016-09-20:
See also:
https://bugs.launchpad.net/hidravfx/+bug/1025262

duro80, 2016-09-20:
Thank You for explanation.

xor-gate, 2016-09-20:
I really appreciate your feedback, I will update the documentation accordingly before the release.

jerrrwalker, 2016-10-03:
in ubuntu-14.04 ldconfig (not ld-config) works ok.
thank you, xor-gate

gvz, 2016-10-17:
Hi,

in Fedora 24 it is also ldconfig. 
Before: same Problem as in original post.
After: works fine

xor-gate, 2016-10-29:
Thanks for reporting ! I have added it to the documentation now.

IdeoG, 2017-06-15:
Hi,
Got same problem. ldconfig helped me.
Thanks

dymbasss, 2018-04-12:
st-flash: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory
---
ldd /usr/local/bin/st-info
	linux-vdso.so.1 =>  (0x00007ffd209cc000)
	libstlink.so.1 => not found
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc398cd5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fc39909f000)
---
ld.so.conf:
# libc default configuration
/usr/local/lib
---
What the problem??? Ubuntu 16.04.

Slavjan, 2018-04-28:
this may be fixed via `sudo ldconfig`

Slavjan, 2018-04-28:
> $ ldd /usr/local/bin/st-util
	linux-vdso.so.1 =>  (0x00007ffe6bad0000)
	libstlink.so.1 => not found
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5ec53f0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5ec57ba000)

> $ sudo ldconfig

> $ ldd /usr/local/bin/st-util
	linux-vdso.so.1 =>  (0x00007ffe6cec3000)
	libstlink.so.1 => /usr/local/lib/libstlink.so.1 (0x00007f54cfc69000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f54cf89f000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f54cf687000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f54cfe7d000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f54d0060000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f54cf46a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f54cf262000)

> $ st-util
st-util 1.4.0-35-gb1b2a2f
2018-04-29T00:27:52 WARN usb.c: Couldn't find any ST-Link/V2 devices

enjoy :)

jiapei100, 2018-05-19:
`sudo ldconfig` solves the problem for me...

eritain, 2019-08-05:
Experience report: The ldconfig fix works if you're installing system-wide. If you install to a user dir, neither ldconfig nor exporting the custom LD_LIBRARY_PATH fixes the problem.

zainabz1995, 2020-02-28:
> Output from ldd /usr/local/st-info:
> 
> > linux-vdso.so.1 => (0x00007ffd4e7f0000)
> > libstlink.so.1 => not found
> > ...
> 
> Solution was:
> 
> > export LD_LIBRARY_PATH=/usr/local/lib
> 
> After this, everythink works fine.
> 
> but I think, this is hack. It can be in installation script or in linker settings.

Yes... Your hack was correct in addition to the solution provided by @xor-gate

Basilisvirus, 2020-03-13:
Hello. sudo ldconfig does not work for me.

When i run 

`ldd /usr/local/bin/st-info`

 i get:
`ldd: /usr/local/bin/st-info: No such file or directory`

in my **/usr/local/lib** there is nothing installed (only python 3.6 folder)

i run `sudo ldconfig` (`sudo ld-config` does not work, and without sudo it will give me **Permission denied**)
But still when i say `st-info --version` , i will get st-info:` error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory`

@xor-gate should i run `make install DESTDIR=_install`  as you suggested above ? but i have already installed the release. Should i reinstall it?

also @duro80 said `export LD_LIBRARY_PATH=/usr/local/lib` worked. Should i run this command? and if yes, should i do so from a specific location in folder?

Thank you

Basilisvirus, 2020-03-13:
Fixed it.
I re-installed the 1.6.0 version, using `sudo make` instead of `make` and it installed correctly, no more errors occured

Nightwalker-87, 2020-03-17:
@Basilisvirus: Could you do me a favour (if you have time and are willing to do so) and try this out with Release v1.3.0 and later? It would be useful to know when this started to work... We currently seem to have no reference on this.

Nightwalker-87, 2020-03-29:
Resolved with commit be66bbf200c718904514b044ba84d64a36456218.

commit be66bbf2 (2016-10-29): doc/compiling.md: Add note about installation and ldconfig. Fixes #478
closed 2020-03-29 by Nightwalker-87