stlink-org/stlink #479: Make the requirement about "pandoc" is configurable

gerryqd, 2016-09-21:
Hi,

I am using stlink for several months,  I buit it on Cygwin, but after recent upgrades, I cannot able to build it out successfully. Seems that "pandoc" is required to generate the manpages.

Is it possilbe to make it configurable? For example, "cmake MANPAGES=NONE".

Or if it cannot find the "pandoc", it will not generate the manpges, just as it checks for the "GTK". 

Thanks & Regards,
Gerry

xor-gate, 2016-09-21:
You are right, currently the Release target will not generate the man-pages with pandoc. The Debug target will. I will patch it so we have cmake option like: `STLINK_ENABLE_DOC` (like nanomsg does: https://github.com/nanomsg/nanomsg/blob/master/CMakeLists.txt#L90-L97)

gerryqd, 2016-09-29:
Seems that it is fixed, right?

I built it on debian 8.6,  though I have not install pandoc, it can be built out successfully.

I also verified this on Cygwin.

texane, 2016-09-29:
Yes. The fix is not clean, but it now works without pandoc installed.

xor-gate, 2016-10-29:
@gerryqd could you try on latest master?

xor-gate, 2016-11-18:
Related to #487, fixed in commit https://github.com/texane/stlink/commit/68b0f3bddc3c4aaffe34caa6a3201029edd8ad56.

closed 2016-11-18 by xor-gate