stlink-org/stlink #482: distribution: Debian packaging is broken yet again

nekromant, 2016-09-22:
Current debian install rules to not take care to install the correct version.h that pretty much make the whole -dev package unusable.

xor-gate, 2016-09-22:
I quote:

> Do we really need to duplicate all the file installation rules in debian/rules? I suggest we do as much of the installation as possible with CMakeLists.txt, so that debian/rules would do as much as possible automatically.
> 
> Is there any rationale for building both debug and release builds of stlink when packaging?

xor-gate, 2016-09-22:
1. I agree. we need to use the install target of cmake.
2. We building a package we should only build release.

xor-gate, 2016-09-22:
> Use of cmake install would be better in terms of maintenance.
> 
> I am not sure how to tell cmake where to install
> adapt debian/*.install files to copy from cmake install dir, instead of from debian/tmp
> or you can manage to install to debain/tmp with cmake
> cmake should handle multiarch library paths
> Just that tests depends on debug build. The build stage calls make all and all depends on release. The tests stage calls make test and test depends on debug. This is why both where build. I think automated tests where a good thing and should run, but you can disable it by add override_dh_auto_tests: to debian/rules, if you want.

xor-gate, 2016-09-22:
You can tell cmake to install into `DESTDIR` with `make install`. See https://cmake.org/cmake/help/v3.0/variable/CMAKE_INSTALL_PREFIX.html

nekromant, 2016-09-22:
@OsterlaD As @texane noted, you can just add DESTDIR= make install. In debian rules you don't have to mention this, as default debian stuff gets the job done already. Basically:
```
override_dh_auto_configure:
        dh_auto_configure -- \
        -DCMAKE_LIBRARY_PATH=$(DEB_HOST_MULTIARCH)
``` 
Is the only thing required and the only thing that I keep around my projects. dpkg does the rest automatically.  
 
@texane For my projects (See http://github.com/nekromant/aura) I add tests for production builds as well, but only those that don't require hardware connected by default. They are just not installed during the 'make install' phase. If dpkg-buildpackage won't find any tests it just won't run any, and since we have a CI I really think we can safely skip tests for packaging.

xor-gate, 2016-09-22:
@nekromant i'm @xor-gate :=) not @texane :D

OsterlaD, 2016-09-22:
I thought about this way too.
dh_auto_build will call `make all`, which calls `cmake` at its own.
This is why dh_auto_configure has no effect.
So I removed code from dh_auto_configure.
There is no `make install`.

xor-gate, 2016-09-22:
We could add `make install`  to the toplevel `Makefile` which just calls `make install` in the release build

OsterlaD, 2016-09-22:
Sounds good. Maybe I have some time to try it later on.

nekromant, 2016-09-22:
@xor-gate I'll give it a proper try tomorrow.

OsterlaD, 2016-09-22:
#484 use of `make install`

Lintian has some suggestions
* rename package to libstlink1, but that conflicts cmake libstlink look-up
* move lib links to dev pkg

I would like to add a postinst script to libstlink to reload udev rules.

xor-gate, 2016-10-09:
I think this is now completely resolved?

nekromant, 2016-10-10:
@xor-gate This is more like optional polishing. Postinst would be nice. 
As for lib links, AFAIK I've done that already.

xor-gate, 2016-10-29:
If you guys don't mind I will close this for now, as this topic was used mostly for discussion. When there specific are debian-related thinks need to be fixed we can open new issues. Thanks all for your effort!

Nightwalker-87, 2020-03-17:
Finally closed by #485.

closed 2020-03-17 by Nightwalker-87