stlink-org/stlink #486: Could not find any ST-Link/V2 devices

BirdBare, 2016-09-24:
- [stlink/v2-onboard ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ Linux] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ v1.2.0-128-g823079a-dirty] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ st-util, st-flash] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
- [stm32f407vg, stm32f767zi ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

I am not able to connect to either of my STM32 devices. I can run the commands from a mac so I know the boards are good.

I ran the cmake and make command exactly as described in the README.
I am also not sure why my version is -dirty.

OUTPUT/ERROR of the commandline tool(s)

bare@bare-linux:~/Desktop/Programming/arm/stlink/src/gdbserver$ ./st-util
st-util 1.2.0-128-g823079a-dirty
2016-09-24T16:15:33 WARN /home/bare/Desktop/Programming/arm/stlink/src/usb.c: Couldn't find any ST-Link/V2 devices

Expected/description:

I expected to upload my program to the device or at least make a connection with st-util. Neither happened on both of my boards.

Thank you

xor-gate, 2016-09-24:
Probably you are not authorized to read/write to the usb devices because the `bare` user has no permissions to do so. You must make sure you usb device has the proper permissions. E.g set them to read/write for a `stlink` group (with udev rules -> see etc folder). And add the `bare` user to it. 

This is just basic linux knowledge, but I have seen different default read/write behaviour on FreeBSD/OpenBSD and Linux. 

Just read here: https://github.com/texane/stlink#setting-up-udev-rules

If there any questions, feel free to ask.

BirdBare, 2016-09-24:
Thank you,
Yeah I was looking for the udev rules. Please update the README file. I did not see it in root so I assumed(my bad) it copied the files there during the make process. I'm sure it will work after I copy the files so I will close this.

BirdBare, 2016-09-24:
Yep it works. Thanks for pointing me to the udev rules

xor-gate, 2016-09-25:
I'm reopening as we need some documentation for users who compile themselves. As most don't know the magic of device nodes, systemd, and udev.

closed 2016-10-22 by xor-gate

nicovv44, 2019-05-10:
As there is no documentation, I still don't know how to do...

commit 9eb84d9f (2016-10-22): README.md: Refactor compilation and udev notes into seperate document. Fixes #486