stlink-org/stlink #487: Build fails on Cygwin,  install Library TARGETS given no DESTINATION!

gerryqd, 2016-09-29:
My environment, Win7 and Cygwin,

I tried to build latest version on Cygwin, and it reports below error,

```
CMake Error at CMakeLists.txt:97 (install):
  install Library TARGETS given no DESTINATION!
```
I searched the web and found below message is useful,

http://stackoverflow.com/questions/14990343/cmake-error-targets-given-no-library-destination-for-shared-library-target/37729574#37729574

I made a fix and it works, the fix is as below,

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 2d5a031..9e39c51 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -94,9 +94,15 @@ else()
        target_link_libraries(${PROJECT_NAME} ${LIBUSB_LIBRARY})
 endif()

+if(CYGWIN)
 install(TARGETS ${PROJECT_NAME}
-   LIBRARY DESTINATION lib/${CMAKE_LIBRARY_PATH}
+    RUNTIME DESTINATION lib/${CMAKE_LIBRARY_PATH}
 )
+else()
+install(TARGETS ${PROJECT_NAME}
+    LIBRARY DESTINATION lib/${CMAKE_LIBRARY_PATH}
+)
+endif()
```

But, I have no idea about "MSYS2" and "MINGW32" cases, as I have no environment to verify it, also, I have no knowledge about the CMake.

The fix needs to be refined, there should exists better way.

xor-gate, 2016-09-30:
Probably this has nothing todo with cygwin, but with different CMake behaviour. I'm not 100% sure. The library should probably not go into the RUNTIME destination. 

Quote from the cmake doc:

> For DLL platforms the DLL part of a shared library is treated as a RUNTIME target and the corresponding import library is treated as an ARCHIVE target. All Windows-based systems including Cygwin are DLL platforms. The ARCHIVE, LIBRARY, RUNTIME, and FRAMEWORK arguments change the type of target to which the subsequent properties apply. If none is given the installation properties apply to all target types. If only one is given then only targets of that type will be installed (which can be used to install just a DLL or just an import library).
> 
> https://cmake.org/cmake/help/v3.0/command/install.html?highlight=install

xor-gate, 2016-10-09:
Could you try with the latest patches in the master branch?

gerryqd, 2016-10-10:
I tried, still report message as below

```
CMake Error at CMakeLists.txt:97 (install):
  install Library TARGETS given no DESTINATION!
```

xor-gate, 2016-10-22:
I will close this issue, as I have tried to fix it in commit https://github.com/texane/stlink/commit/68b0f3bddc3c4aaffe34caa6a3201029edd8ad56 and this is a duplicate of a newer bug #506 (as we can discus further there). Thanks!

closed 2016-10-22 by xor-gate