stlink-org/stlink #488: make is not build st-util

kv193, 2016-09-29:
- [Stlink/v2-onboard ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [Linux-OpenSUSE Leap 64-bit ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)

I do all step in Readme. But after make st-util not built (output is attached)
[out.txt](https://github.com/texane/stlink/files/501301/out.txt)
What I do wrong?

closed 2016-09-29 by kv193

xor-gate, 2016-09-29:
Which version are you compiling? Have you tried with latest master?

aguegu, 2016-10-25:
I have the same problem on the latest master branch, 1.3.0-dev on ubuntu. Do not know why this issue is closed.

xor-gate, 2016-10-25:
Please open a new issue, with the details of your build and system.

Nightwalker-87, 2020-03-17:
`st-util` is built and placed in the subdirectory `src/gdbserver/st-util`. See also #508. 
This issue is now permanently closed.