stlink-org/stlink #490: Case: while using use GPIOA port jtag tools not work. What can I do?

kv193, 2016-09-30:
- [Stlink/v2-onboard ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ V2J27M15] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ OpenSUSE 13.2] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ v1.2.0/git-9daea3c273] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ st-flash] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
- [STM32F411RE (Nucleo-F411RE) ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

Hello.
I am novice in STM32. But I read all issue carefully and after many attempts to solve my problem read my result. So I adapt 1-st blinky program from STM32-Discovery to my Nucleo-F411RE. I use stlink for burn flash. And 1-st time was success. But no more times, every time I got result some like:

```
blinky > make burn
/home/kv193/Programs/arm/stlink/build/st-flash --reset write blinky.bin 0x8000000
st-flash 1.2.0-129-gee3f7a1
2016-09-30T18:18:34 INFO /home/kv193/Programs/arm/stlink/src/common.c: Loading device parameters....
2016-09-30T18:18:34 WARN /home/kv193/Programs/arm/stlink/src/common.c: unknown chip id! 0xa05f0000
Makefile:51: recipe for target 'burn' failed
make: *** [burn] Error 255
```
After spent time I find decision: . use STM ST-LINK Utility for Windows.
This utility success erase my chip ONLY IN SWD connection settings with mode "Connect Under Reset". In JTAG mode also I had error. After reading documentation I notice that used GPIOA por is include JTAG pins. And result if I config "RCC->AHB1ENR |= RCC_AHB1ENR_GPIOAEN;" then JTAG not work, but SWD work fine. Is it true? And if "yes" how to use SWD in Linux?

xor-gate, 2016-10-03:
When you program the STM32F411RE on the nucleo with the onboard programmer there should be no problem. However when you use the SWD pins in your firmware it should also work as `texane/stlink` will also connect under reset and those pins are available under reset.

When you have access to a linux machine you cloud try openocd: `openocd -f <conffile>` with conffile:

```
source [find interface/stlink-v2.cfg]
transport select hla_swd
source [find target/stm32f4.cfg]
reset_config srst_only
```

xor-gate, 2016-10-29:
Closing due to inactivity, feel free to comment.

closed 2016-10-29 by xor-gate

kv193, 2017-03-20:
Excuse me for long time silence (I work on other project with Cortex-M3 from diff manufacturer ).
Now I return to study STM32. And I continue ... Without / with reset I get different errors (see attachments). What is my wrong?
[console.txt](https://github.com/texane/stlink/files/854974/console.txt)

xor-gate, 2017-03-20:
You should try a least the newer released v1.3.1.

kv193, 2017-03-21:
Yes, I am already do it.The ver 1.3.1work fine.Thank you. -- Best wishesKlochkov V. A.email: kv193@yandex.rumob: +7-911-9141440

xor-gate, 2017-03-21:
@kv193 thank you for your feedback and testing.