stlink-org/stlink #492: Questions about st link

gandol2, 2016-10-04:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropiate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropiate.

- [ ] Programmer/board type: Stlink/v2-onboard
- [ ] Programmer firmware version: lost
- [ ] Operating system: Windows 10
- [ ] Stlink tools version and/or git commit hash: 
- [ ] Stlink commandline tool name: 
- [ ] Target chip (and optional board): STM32F401RE

Hi. 
I lost st-link firmware of NUCLEO-F401RE. 
I can not download the target device(STM32F401) firmware.
I need st-link firmware version V2.Jxx.Mxx..
Can I obtain a st link V2-1 hex file?

xor-gate, 2016-10-04:
I must say this is not related to our project. You should use the original firmware upgrade tool from STlink itself or else you can move your programmer to the trashcan. Firmware on the program "doesn't get lost"  by itself (normally). If you can connect to the programmer with the STLink firmware upgrade tool or STUtility from ST itself then you are out of luck.

gandol2, 2016-10-04:
Sorry yiraseo questions unrelated to the current project.
After various attempts, and currently it operates V2.Jxx.Sxx version.
Should it not possible to change it back to V2.Jxx.Mxx version?
Sorry again yiraseo questions not related to the project .. T.T

xor-gate, 2016-10-04:
I don't understand what you say. Why should you change back the firmware? I have never heard of Mxx version, as only Sxx (SWIM interface) is available.

gandol2, 2016-10-04:
Nucleo have Mxx version.
Disco and ST-LINK/V2 have Sxx version.
I have an Sxx version hex file downloaded from Google.
but I really need Mxx version hex file..

xor-gate, 2016-10-09:
I think you are at the wrong place for this question, I would like to help but I have no idea how you broke it.

xor-gate, 2016-10-15:
@gandol2 I have found an article about firmware corruption:
http://www.carminenoviello.com/2016/02/26/restore-st-link-interface-bad-update-2-26-15-firmware/

closed 2016-10-22 by xor-gate