stlink-org/stlink #493: what can be a problem with?

night-ghost, 2016-10-04:
Sometimes after reflash via ST-Link STM32F4 (05RGT6 in my case) turns to strange state and start to do commands wrong. [video](https://youtu.be/YYp3GQnN-HA) 

reset can not deduce from this state, only the power off. It is definitely not a problem of the Eclipse - Keil doing the same. It is also unlikely that it is a failure of the CPU - ARM quite licked. But what it can be?

PS. Can st-link do a power cycle for connected board if it powered from st_link?

xor-gate, 2016-10-09:
I'm not sure what causes the problem, you are the first one reporting this behaviour. I dislike you didn't fill in the [full issue report](https://github.com/texane/stlink/blob/master/.github/ISSUE_TEMPLATE.md) as I can't know which stlink version you are using.

When you have access to a linux machine you cloud try openocd: `openocd -f <conffile>` with conffile and see if it works more stable:

```
source [find interface/stlink-v2.cfg]
transport select hla_swd
source [find target/stm32f4.cfg]
reset_config srst_only
```

night-ghost, 2016-10-14:
Thanks for answer. I use chinese ST-Link v2 from Ali so I don't know details about it. This issue is rather rare (4 times per 2 weeks), and fixed by power cycling, so let it stay, not a serious obstacle.

xor-gate, 2016-10-22:
Thats a pain, the project cannot guarantee stability for the stlink clones. The nucleo or discovery boards and the programmers itself are so cheap there is no excuse to have chinese crap.

night-ghost, 2016-10-24:
When discussing the price questions with the representatives of other countries, multiply in mind the price of 100 - it will not go wrong. May be nucleo is very cheap in dollars - but just the dollar is very expensive.

Nightwalker-87, 2020-03-17:
Reopening this as this does not seem to be a satisfiable result.

Nightwalker-87, 2020-06-03:
There are no issues with Chinese STLINK-v2 clones in general - they usually work well with the `stlink` toolset - even with recent original ST firmware versions. The mentioned problem could also be related to faulty hardware.

The linked video by @night-ghost is not helpful due to the very low resolution and we don't know the exact hardware that was used by then which leaves us without any possibility to reproduce.

Unless there is no further feedback from the original issuer this ticket will be closed in 7 days.

Nightwalker-87, 2020-07-19:
Closing this as there is no chance for any further investigation.

closed 2020-07-19 by Nightwalker-87