stlink-org/stlink #494: Incorrect SN

luckyluke72, 2016-10-04:
Programmer/board type: Stlink/v2
Programmer firmware version: NA
Operating system: Linux
Stlink tools version and/or git commit hash: v1.2.0-129-gee3f7a1
Stlink commandline tool name: st-info --serial
Target chip (and optional board): ANY

IMHO, there is an error in stlink_print_serial(stlink_t *sl, bool openocd), file src/tools/info.c:22

line 30: fmt = "%02x";

should be replaced with 

fmt="%c";

then  I noticed the SN is truncated. In facts, the ST SN is 24 chars long. You have to extend the serial number array from 12 to 24 chars in some places in your code. In fact the real SN of the board should have been

0667 FF53 5651 7270 6761 3012

See for usb.c:912 stlink_probe_usb_devs(libusb_device *devs, stlink_t *sldevs[]) and then in line 967 it should be handled with a different size of serial

usb.c:958 char serial[13];

and include/st-link.h:135

       char serial[25];
 
Then it works

[ mar ott 04  14:45:03 ] celettol@celettol-HP-Compaq-Elite-8300-USDT:~/software/stlink/build$ ./st-info --serial
0667FF535651727067173012
[

xor-gate, 2016-10-09:
The nucleo board programmer (stlinkv2-1) formats the serial in a hex-encoded string format (usb pid 0x374b). While stlink v1 and v2  format it is binary. It is already known this is incompatible, it needs more work than just changing the array to 25 elements. 

Thank you for your contribution, as this is a duplicate of #417 I will close this. Feel free to discus further in the other issue. Thanks!

tarek-bochkati, 2021-03-16:
#1106

closed 2021-03-20 by Nightwalker-87