stlink-org/stlink #498: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

smartHarsh, 2016-10-13:
Performing C SOURCE FILE Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS failed with the following output:
Change Dir: /home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_309db/fast"
/usr/bin/make -f CMakeFiles/cmTC_309db.dir/build.make CMakeFiles/cmTC_309db.dir/build
make[1]: Entering directory '/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_309db.dir/src.c.o
/usr/bin/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS   -Wmissing-variable-declarations -o CMakeFiles/cmTC_309db.dir/src.c.o   -c /home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-Wmissing-variable-declarations'
CMakeFiles/cmTC_309db.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_309db.dir/src.c.o' failed
make[1]: *** [CMakeFiles/cmTC_309db.dir/src.c.o] Error 1
make[1]: Leaving directory '/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_309db/fast' failed
make: *** [cmTC_309db/fast] Error 2

Source file was:
int main(void) { return 0; }
Performing C SOURCE FILE Test C_SUPPORTS_WSHORTEN_64_TO_32 failed with the following output:
Change Dir: /home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_83ce9/fast"
/usr/bin/make -f CMakeFiles/cmTC_83ce9.dir/build.make CMakeFiles/cmTC_83ce9.dir/build
make[1]: Entering directory '/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp'
Content of Error.log file was:-
Building C object CMakeFiles/cmTC_83ce9.dir/src.c.o
/usr/bin/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WSHORTEN_64_TO_32   -Wshorten-64-to-32 -o CMakeFiles/cmTC_83ce9.dir/src.c.o   -c /home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-Wshorten-64-to-32'
CMakeFiles/cmTC_83ce9.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_83ce9.dir/src.c.o' failed
make[1]: *** [CMakeFiles/cmTC_83ce9.dir/src.c.o] Error 1
make[1]: Leaving directory '/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_83ce9/fast' failed
make: *** [cmTC_83ce9/fast] Error 2

xor-gate, 2016-10-13:
This means your compiler doesn't support `-Wmissing-variable-declarations` and is not a real issue. 

https://github.com/texane/stlink/blob/master/cmake/CFlags.cmake#L27

closed 2016-10-13 by xor-gate

smartHarsh, 2016-10-13:
Sir but I am not able to build on providing command "make" it gives "make: *** No targets specified and no makefile found.  Stop."

xor-gate, 2016-10-13:
You should run `make clean` and then `make` without aborting the make.

smartHarsh, 2016-10-13:
Sir , There is no makefile generated by cmake.
cmake .. gives the error:-

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/gdbserver
   used as include directory in directory /home/harsh/Applications/ARMToolchain/stlink-master/src/tools/gui
[... cut ...]
    linked by target "stlink-static" in directory /home/harsh/Applications/ARMToolchain/stlink-master
    linked by target "stlink" in directory /home/harsh/Applications/ARMToolchain/stlink-master
    linked by target "stlink" in directory /home/harsh/Applications/ARMToolchain/stlink-master

-- Configuring incomplete, errors occurred!
See also "/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/harsh/Applications/ARMToolchain/stlink-master/build/CMakeFiles/CMakeError.log".

xor-gate, 2016-10-13:
You are missing the libusb dependency, you have not filled in the [issue template](https://github.com/texane/stlink/blob/master/.github/ISSUE_TEMPLATE.md) so I can not help you completely because of missing details.

smartHarsh, 2016-10-14:
I am using it to configure stm32f100rb with  GNU ARM eclipse plugins , st-linkv2 and STM32F1 Discovery board.

smartHarsh, 2016-10-14:
And is using ubuntu operating system

xor-gate, 2016-10-14:
Before compiling `texane/stlink` you should install `libusb-1.0.0-dev`, as stated in the README here: https://github.com/texane/stlink#common-requirements

smartHarsh, 2016-10-14:
Thanks friend