stlink-org/stlink #499: Issues understanding the st link install for Mac

moha1899, 2016-10-15:
I guess I'm just really new to this kind of stuff and I'm just having a hard time understanding exactly how to install the st link for a MacBook that's On Yosemite 10.10.5. I was wondering if you could possibly send me a more detailed step by step on how to install and what files need to be changed and stuff. I'd really appreciate it. I look forward to hearing back from you.

xor-gate, 2016-10-15:
I totaly understand, as a user you should use Homebrew package manager (installing opensource on your mac). When you want the latest software you need to compile yourself which is wat is described in this README.md of the project.

I think you should go this route:

* Install [iterm2](https://www.iterm2.com/) terminal (better than console)
* Install [homebrew package manager](http://brew.sh) (like aptitude on ubuntu/debian, or chocolaty on windows)
* Install compiler toolchain and programming tools for ARM
  * `brew install stlink`
  * `brew install openocd` (alternative for `texane/stlink`)
  * `brew install Caskroom/cask/gcc-arm-embedded`

You are right, this is not mentioned in the readme for first-time users. I have labeled this issue It should land into the documentation.

xor-gate, 2016-10-15:
I could recommend this book https://leanpub.com/mastering-stm32

moha1899, 2016-10-15:
I definitely will it. I installed the opencd but now I'm kind of stuck running it. Are those the exact steps I need to run it? Thank you so much for clearing things up and

Abdulqadir Mohamud

xor-gate, 2016-10-15:
You need to write the configuration into a file e.g `myopenocd.cfg` and then run from the commandline: `openocd -f myopenocd.cfg`. You should see something like:

```
Jerrys-MBP-405:lesson-4 jerry$ openocd -f openocd.cfg
Open On-Chip Debugger 0.9.0 (2016-09-30-17:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
adapter speed: 4000 kHz
trst_and_srst trst_pulls_srst srst_nogate trst_push_pull srst_open_drain connect_assert_srst
Info : clock speed 4000 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v0 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.920499
Info : stm32l1.cpu: hardware has 6 breakpoints, 4 watchpoints
```

xor-gate, 2016-10-22:
How is the progress going?

closed 2016-10-29 by xor-gate