stlink-org/stlink #500: support for stm32l432 nucleo

geoffreymbrown, 2016-10-15:
Yet another new chip.  I've figured out how to add support to st-util, but before I do a pull request, I'd like some feedback.  

The chip id is 0x435

It looks like this is used for other parts (from an emblocks forum)

STM32L431CC Device ID 0x435
STM32L431KC Device ID 0x435
STM32L432KC Device ID 0x435

and from RM0349  for  STM32L431/433/443

So my question is, what should the new chip id constant be called ?  I started to call it

STLINK_CHIPID_STM32_L43

but that doesn't seem right.

**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropiate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropiate.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2016-10-15:
Thank you for your contribution. I think we should name it `STLINK_CHIPID_STM32_L43X`. It is indeed DeviceID `0x435`.

xor-gate, 2016-10-15:
If you create a pull request we can review your work.

geoffreymbrown, 2016-10-15:
Ok,  I hope I did it correctly.  This is a new process for me.

Geoffrey

xor-gate, 2016-10-22:
It has been merged and closing this now. Thanks!

closed 2016-10-22 by xor-gate