stlink-org/stlink #504: STM32L151 write half pages failed with -1,  even after mass erase

afnid, 2016-10-16:
# Problem
st-flash 1.2.0-134-gf80412c
2016-10-16T09:12:13 INFO common.c: Loading device parameters....
2016-10-16T09:12:13 INFO common.c: Device connected is: L1 Med-density device, id 0x10386416
2016-10-16T09:12:13 INFO common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 256 bytes
2016-10-16T09:12:13 INFO common.c: Attempting to write 8004 (0x1f44) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001f00 erased
2016-10-16T09:12:13 INFO common.c: Finished erasing 32 pages of 256 (0x100) bytes
2016-10-16T09:12:13 INFO common.c: Starting Half page flash write for STM32L core id
2016-10-16T09:12:13 INFO flash_loader.c: Successfully loaded flash loader in sram
2016-10-16T09:12:22 ERROR flash_loader.c: flash loader run error
2016-10-16T09:12:22 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2016-10-16T09:12:22 WARN common.c: 
write_half_pages failed == -1

# Checklist

- [x] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard

STML151R8T6 production board using external ST Micro ST-LINK/V2 (fails)
STML152 Nucleo board using on-board programmer (works)

- [x] Programmer firmware version: e.g STSW-LINK007 2.27.15

Originally with what was already there, but both have been upgraded to latest:
STLinkUpgrade 3.2.0
Firmware:V2j2856

- [x] Operating system: e.g Linux, Mac OS X, Windows (with specific version)

Two different computers with latest Ubuntu Mint
Tested on Mac OS Darwin 15.6.0

- [x] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056

Versions:
On linux, a previous snapshot plus latest git pull: v1.2.0-134-gf80412c
Didn't bother to upgrade the mac: v1.2.0-127-g30523ed

- [x] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 

st-flash

- [x] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

As stated, and shown in logs below

A as-detailed description possible of the problem with debug output when available.

# Output:

**Flash L151:**

: st-flash erase
st-flash 1.2.0-134-gf80412c
2016-10-16T09:41:10 INFO common.c: Loading device parameters....
2016-10-16T09:41:10 INFO common.c: Device connected is: L1 Med-density device, id 0x10386416
2016-10-16T09:41:10 INFO common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 256 bytes
-> Flash page at     0/  512 erased
-> Flash page at     1/  512 erased
-> Flash page at     2/  512 erased
-> Flash page at     3/  512 erased
-> Flash page at     4/  512 erased
-> Flash page at     5/  512 erased
-> Flash page at     6/  512 erased
-> Flash page at     7/  512 erased
-> Flash page at     8/  512 erased
-> Flash page at     9/  512 erased
-> Flash page at    10/  512 erased
-> Flash page at    11/  512 erased
[... 209 lines cut ...]
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 4 0x4
 00 00 00 00

# Expected/description:

I modified some of the log messages to compare to a working Nucleo, but don't see any significant differences.

I have been able to do a st-flash erase several times, and a st-flash read generates a file full of 0's as expected, but didn't do a read before the first erase.

I have custom boards I have jumper wires direct from the pins on the programmer to a header on the pcb.  Power has been provided through the st-link and independently with identical results for both erase and write.

Unfortunately flashing these boards is on my critical path, so you have my attention as long as you need it.

I have a L152 discovery and some other series nucleo/discovery boards if you would like me to try another test configuration.

xor-gate, 2016-10-16:
You should try the latest `master` branch.

xor-gate, 2016-10-16:
I had do read over the verbose log, i'm not yet sure what causes the problem.

afnid, 2016-10-18:
Can we narrow down where the problem is or isn't?  Definitely not an expert with this so need to know where to direct my time.

I am assuming a successful bulk erase ruled out any JTAG wiring/connection issues? 

Same issue across platforms, so could rule out the usb libraries unless there is a chip specific race condition?

Known issue, so not caused by a regression in the source code.  The erase should have worked, but I have also tried with boot0 pulled high, so can rule out any target chip code execution.

Assuming a bulk erase can't bypass the chip in some way, does this narrow down to either a bug with the st-link programmer firmware or something is broke/missing in st-util and this series of processor?

Does that snippet of assembly run in the ram of the st-link or in the ram of the target chip?  Are there any reference docs on any of the st-link internals?

I noticed a different register used to id the lower density chip, but didn't appear to be that significant.  Maybe some other register offset issues not ruled out by a successful mass erase?

I have the source to the original boot-loader that was flashed on the chip.  Unless the std peripheral lib was modified, it isn't anything special.

Any hunches?

xor-gate, 2016-10-18:
The protocol of the stlink is under NDA (reversed engineered). The flash loader runs in the microcontroller in SRAM and is controlled by `texane/stlink` with commands. I don't know why you closed it already. I have a `stm32l152rct6` discovery and it works without problems.

afnid, 2016-10-18:
oops, didn't mean to do that.

afnid, 2016-10-19:
The stm32l151 uses the medium density profile which was the one thing I saw in common with issue #356 where the half-page writes failed.  The stm32l152rct6 uses the medium plus chip profile.

Since the medium profile uses 0x1ff8004c for the size register and the other profiles use 0x1ff800cc, maybe there are some other differences?

afnid, 2016-10-19:
This time I am closing intentionally.

Solution was to connect pin 1 to target chip vcc, then jumper pin 19 to pin 2 to power the chip.  It appears the programmer has to be able to sense voltage on the target chip through pin 1/2.

Seemed logical to connect pin 19 direct to my target vcc since it was the only one with power.  There are image search results that are incorrect that I may have used, but one leads to a post that explained this.

[image]

closed 2016-10-19 by afnid