stlink-org/stlink #506: Compilation for Windows (cywin) troubles

LonelyWolf, 2016-10-17:
Tried to compile the version from 16.oct.2016 for Windows (cygwin), it have major changes in compilation process since the last one I've tried on Windows. So finally got all compiled but had to "fix" the following issues:

- **First issue**:
It required a pandoc.
```
CMake Warning at cmake/modules/pandocology.cmake:171 (message):
  Pandoc not found.  Install Pandoc (http://johnmacfarlane.net/pandoc/) or
  set cache variable PANDOC_EXECUTABLE.
Call Stack (most recent call first):
  doc/man/CMakeLists.txt:13 (add_document)
```
Well, I've installed a windows version of this utility, that helped (but in fact that was useless, see issue#3).

- **Second issue**:
```
CMake Error at CMakeLists.txt:97 (install):
  install Library TARGETS given no DESTINATION!
```
I've changed the `CMakeLists.txt`:
from:
```
install(TARGETS ${PROJECT_NAME}
    LIBRARY DESTINATION lib/${CMAKE_LIBRARY_PATH}
)
```
to:
```
install(TARGETS ${PROJECT_NAME}
    DESTINATION lib/${CMAKE_LIBRARY_PATH}
)
```
The `cmake` finished with success.

Then I've run the `make` and got
- **Third issue**:
```
[ 59%] Generating st-term.1
pandoc.exe: /cygdrive/r/stlink/build/doc/man/st-term.md: openFile: does not
exist (No such file or directory)
```
The file `st-term.md` (and others `.md`) does exist, but windows version of
pandoc can't see it because of unix-style path. If I run manually ...
```
pandoc.exe r:\stlink\build\doc\man\st-term.md
```
... all be fine. In cygwin path `/cygdrive/r/stlink/build/doc/man/st-term.md` is equal to normal windows path `r:\stlink\build\doc\man\st-term.md`.
I'm not very familiar with `cmake` to fix such path-generation for `pandoc` on windows, so I've
removed a string:
```
"add_subdirectory(doc/man)"
```
from `CMakeLists.txt` and finally all stlink utilities were compiled.

Hope this information will be useful to fix issues in windows compilation.

xor-gate, 2016-10-22:
Hi @LonelyWolf sorry for my late reply. I have pushed changes to the master branch which you can test out. Hopefully I have not broken other builds (mac/linux/unix), but we will see. Could you test this out:

https://github.com/texane/stlink/commit/68b0f3bddc3c4aaffe34caa6a3201029edd8ad56

LonelyWolf, 2016-10-22:
Got the latest sources from repo, then:
```
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug ..
make
```
Compilation success.
Thanks.

LonelyWolf, 2016-10-22:
Also I can confirm what this tool flawlessly works with STM32L151RD and STM32L476RG via:
- ST-Link v1 (from the old STM32VLDISCOVERY board)
- ST-Link v2 (the ST programmer)
- ST-Link v2.1 (Nucleo board)
This is for `known working targets`.

closed 2016-10-24 by LonelyWolf

xor-gate, 2016-10-24:
Great! Thanks for you report.