stlink-org/stlink #508: st-util is not built in the latest master branch

aguegu, 2016-10-26:
In the latest version in master branch, as 1.3.0-dev, st-util is not built, according to the compile document, referring to #488 

[image]
[image]

My system is ubuntu 16.10. 

Well, all the stdout log seems all right, but the result is missing.

xor-gate, 2016-10-26:
It is build but the documentation doesn't say anything where the executable is located. You should install it from the build folder and run from there like this (not sure it is installed in /usr/local:

```
make install DESTDIR=_install
./_install/usr/local/bin/st-util ....
```

aguegu, 2016-10-27:
[image]

I ran the `make` in build folder, the bins are not in ./usr, neither /usr/local, where require root privilege.

xor-gate, 2016-10-27:
```
Jerrys-MBP-405:Debug jerry$ ls
CMakeCache.txt		_install		libstlink.a
CMakeFiles		cmake_install.cmake	libstlink.dylib
CPackConfig.cmake	dist			src
CPackSourceConfig.cmake	doc			st-flash
CTestTestfile.cmake	include			st-info
DartConfiguration.tcl	install_manifest.txt	tests
Makefile		libstlink.1.2.0.dylib	usr
Testing			libstlink.1.dylib
Jerrys-MBP-405:Debug jerry$ make install DESTDIR=_install
[ 23%] Built target stlink
[ 30%] Built target st-info
[ 53%] Built target stlink-static
[ 63%] Built target st-flash
[ 76%] Built target st-util
[ 86%] Built target flash
[ 93%] Built target usb
[100%] Built target sg
Install the project...
-- Install configuration: "Debug"
-- Up-to-date: _install/usr/local/lib/libstlink.1.2.0.dylib
-- Up-to-date: _install/usr/local/lib/libstlink.1.dylib
-- Up-to-date: _install/usr/local/lib/libstlink.dylib
-- Installing: _install/usr/local/lib/libstlink.a
-- Installing: _install/usr/local/bin/st-flash
-- Installing: _install/usr/local/bin/st-info
-- Installing: _install/usr/local/bin/st-util
-- Up-to-date: _install/usr/local/lib/pkgconfig/stlink.pc
-- Up-to-date: _install/usr/local/include/stlink.h
-- Up-to-date: _install/usr/local/include/stlink/backend.h
-- Up-to-date: _install/usr/local/include/stlink/chipid.h
-- Up-to-date: _install/usr/local/include/stlink/commands.h
-- Up-to-date: _install/usr/local/include/stlink/flash_loader.h
-- Up-to-date: _install/usr/local/include/stlink/logging.h
-- Up-to-date: _install/usr/local/include/stlink/mmap.h
-- Up-to-date: _install/usr/local/include/stlink/reg.h
-- Up-to-date: _install/usr/local/include/stlink/sg.h
-- Up-to-date: _install/usr/local/include/stlink/usb.h
-- Up-to-date: _install/usr/local/include/stlink/version.h
-- Up-to-date: _install/usr/local/share/man/man1/st-util.1
-- Up-to-date: _install/usr/local/share/man/man1/st-flash.1
-- Up-to-date: _install/usr/local/share/man/man1/st-info.1
```

xor-gate, 2016-10-27:
I'm not sure, but you should get the same install list after running `make install DESTDIR=_install`

aguegu, 2016-10-27:
[image]
[image]

Well, it is interesting. After executing `make install DESTDIR=_install` in `bulid` directory, st-util showed up. 

The problem is that `st-flash`, `st-info` are built when `make`, but not `st-util`.

xor-gate, 2016-10-29:
It is build actually, but it is located at `src/gdb-server/st-util` after build.

closed 2016-10-29 by xor-gate

aguegu, 2016-10-29:
Great, I think the project README needs updates too, 

https://github.com/texane/stlink/blob/master/README.md#using-the-gdb-server