stlink-org/stlink #509: STM32F767ZI: unknown chip id! 0xa05f0000

sergiuszm, 2016-10-27:
- [ ] Programmer/board type: Stlink/v2-onboard
- [ ] Programmer firmware version: e.g V2J27M15
- [ ] Operating system: Mac OS X
- [ ] Stlink tools version and/or git commit hash: v1.2.0 https://github.com/texane/stlink/commit/62fa371b421ce929b216a2265931f9c1546d1a69
- [ ] Stlink commandline tool name: `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F767ZI

A as-detailed description possible of the problem with debug output when available.

Output:

st-flash
```
2016-10-27T14:10:55 INFO /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: Loading device parameters....
2016-10-27T14:10:55 WARN /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: unknown chip id! 0xa05f0000
```

st-util
```
2016-10-27T14:13:29 INFO /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: Loading device parameters....
2016-10-27T14:13:29 WARN /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: unknown chip id! 0x2010000
```

Expected/description:
`I should be able to flash board or at least get the board recognised by the st-link toolkit`

xor-gate, 2016-10-29:
You should try the master branch (as this commit is done https://github.com/texane/stlink/commit/3daa3f15cc5f3fe6f33f4bc33b3b61dbcc0c6ce7)

sergiuszm, 2016-10-30:
@xor-gate Thanks, now it works.

`st-util 1.2.0-144-g9c05837
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/common.c: Loading device parameters....
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/common.c: Device connected is: F76xxx device, id 0x10006451
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/common.c: SRAM size: 0x80000 bytes (512 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: Chip ID is 00000451, Core ID is  5ba02477.
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: Chip clidr: 09000003, I-Cache: off, D-Cache: off
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c:  cache: LoUU: 1, LoC: 1, LoUIS: 0
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c:  cache: ctr: 8303c003, DminLine: 32 bytes, IminLine: 32 bytes
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: D-Cache L0: 2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: f00fe019 LineSize: 8, ways: 4, sets: 128 (width: 12)
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: I-Cache L0: 2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: f01fe009 LineSize: 8, ways: 2, sets: 256 (width: 13)
2016-10-30T18:47:27 INFO /Users/serchio/Documents/Projects/stlink/src/gdbserver/gdb-server.c: Listening at *:4242...`

Now I just need to wait for a new release to get it working with homebrew.

closed 2016-10-30 by sergiuszm

xor-gate, 2016-10-31:
@sergiuszm great!