stlink-org/stlink #510: Wrong extract command in FindLibUSB.cmake

slyshykO, 2016-11-05:
diff --git a/cmake/modules/FindLibUSB.cmake b/cmake/modules/FindLibUSB.cmake
index 45472b9..84020b9 100644
--- a/cmake/modules/FindLibUSB.cmake
+++ b/cmake/modules/FindLibUSB.cmake
@@ -24,7 +24,7 @@ if(WIN32 OR CMAKE_VS_PLATFORM_NAME OR MINGW OR MSYS)
                )
        endif()
        file(MAKE_DIRECTORY ${LIBUSB_WIN_OUTPUT_FOLDER})
-       execute_process(COMMAND ${ZIP_LOCATION} x -y ${LIBUSB_WIN_ARCHIVE_PATH} -o${LIBUSB_WIN_OUTPUT_FOLDER})
+       execute_process(COMMAND ${ZIP_EXECUTABLE} x -y ${LIBUSB_WIN_ARCHIVE_PATH} -o${LIBUSB_WIN_OUTPUT_FOLDER})
 endif()

 # FreeBSD

xor-gate, 2016-11-07:
That a good catch, could you create a pull request so you get all the credits? 👍

slyshykO, 2016-11-07:
I made pull request - https://github.com/texane/stlink/pull/511

closed 2020-03-17 by Nightwalker-87