stlink-org/stlink #512: STM32L082KZ Support

tsteinholz, 2016-11-18:
When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [x] Programmer/board type: Stlink/v2
- [x] Programmer firmware version: Latest, just updated with the offical updater on windows.
- [x] Operating system: Mac OS X
- [x] Stlink tools version: "Unavailable", latest on homebrew.
- [x] Stlink commandline tool name: `st-info`, `st-flash`, `st-util`
- [x] Target chip (and optional board): STM32L082KZ

A as-detailed description possible of the problem with debug output when available.

Output:

```
2016-11-18T11:19:37 INFO /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: Loading device parameters....
2016-11-18T11:19:37 WARN /tmp/stlink-20161001-4407-8f1gha/stlink-1.2.0/src/stlink-common.c: unknown chip id! 0x20086447```
```

Expected/description:
`to connect to the chip`

Thank you

xor-gate, 2016-11-18:
Could you try with a unreleased version (master branch of github) which you compile yourself? 

https://github.com/texane/stlink/blob/master/doc/compiling.md#mac-os-x

tsteinholz, 2016-11-18:
@xor-gate yes I can give that a try - then I will update you. Thanks.

tsteinholz, 2016-11-18:
@xor-gate So I only see teh st-flash and st-info binaries. But before when I ran the st-info executable, it never returned anything. And now it does. So that's a good sign, where can I find the st-util binary?

tsteinholz, 2016-11-18:
@xor-gate Here's the updated, with the latest build.

```
st-util 1.2.0-146-g470fc2b
2016-11-18T11:47:23 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Loading device parameters....
2016-11-18T11:47:23 WARN /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: unknown chip id! 0xa05f0000
```

tsteinholz, 2016-11-18:
It might be useful to update homebrew 👍

xor-gate, 2016-11-18:
Homebrew uses only releases and no development snapshots. Some weeks ago I started preparing release v1.3.0 but has not happend yet. In the meantime you can use your own build version. I have no due date yet when v1.3.0 is ready (hopefully before end this year).

Thank you for your response!

closed 2016-11-18 by xor-gate