stlink-org/stlink #513: Extremely Slow loading from GDB

tsteinholz, 2016-11-18:
I have been getting some very strange behavior when trying to load my firmware onto my chip. I have been able to write to flash (0x8000000) on the ST Link Utility (Windows), the 'st-flash' program (Mac OS X), and now I am working on trying to get it to flash via the gdbserver (st-utility). I was originally having issues with my linker script since I could not find an official one for STM32L082KZ architecture. I was only supplied with mdk-keil project files (which don't work on Mac OS X nor do I want to purchase keil) which I ported to arm-none-eabi. Everything is compiling and the linker might be giving some wrong information.

```
/* Linker script to configure memory regions. */
MEMORY
{
  FLASH (rx)  : ORIGIN = 0x08000000, LENGTH = 512K
  RAM   (rwx) : ORIGIN = 0x20000000, LENGTH = 256K
}

/* Library configurations */
GROUP(libgcc.a libc.a libm.a libnosys.a)

/* Linker script to place sections and symbol values. Should be used together
 * with other linker script that defines memory regions FLASH and RAM.
 * It references following symbols, which must be defined in code:
 *   Reset_Handler : Entry of reset handler
 *
 * It defines following symbols, which code can use without definition:
 *  Original                   Port
 * --------------------------------------------------------------------
 *   __exidx_start
 *   __exidx_end
 *   __copy_table_start__
 *   __copy_table_end__
 *   __zero_table_start__
 *   __zero_table_end__
 *   __etext                -> _estack [guess]
 *   __data_start__         -> _sdata
 *   __preinit_array_start
 *   __preinit_array_end
 *   __init_array_start     -> _sidata [guess]
 *   __init_array_end
 *   __fini_array_start
 *   __fini_array_end
 *   __data_end__           -> _edata
 *   __bss_start__          -> _sbss
 *   __bss_end__            -> _ebss
 *   __end__
 *   end
 *   __HeapBase
 *   __HeapLimit
 *   __StackLimit
 *   __StackTop
 *   __stack
 *   __Vectors_End
 *   __Vectors_Size
 */
ENTRY(Reset_Handler)
``` 

When porting the linker file to the arm-none-eabi platform I had to change some of the definitions to match the startup assembly file variables, and I also set the flash and ram origins. Originally I had set the flash origin to 0x0 (this instantly failed), but now I have the origin set to 0x08000000 which causes this strange behavior. Everything else in this linker file should work for a Cortex-M0plus processor.

- [x] Programmer/board type: Stlink/v2
- [x] Programmer firmware version: Latest updated with official tools yesterday.
- [x] Operating system: Mac OS X
- [x] Stlink tools version and/or git commit hash: master 470fc2b309be91b7e0ee9e01bfb60128ec5ae5be
- [x] Stlink commandline tool name: `st-util`
- [x] Target chip (and optional board): STM32L082KZ

Output:
```
2016-11-18T13:37:49 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Attempting to write 128 (0x80) bytes to stm32 address: 134239360 (0x8005480)
Flash page at addr: 0x08005480 erased
2016-11-18T13:37:49 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Finished erasing 1 pages of 128 (0x80) bytes
2016-11-18T13:37:49 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Starting Half page flash write for STM32L core id
2016-11-18T13:37:49 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/flash_loader.c: Successfully loaded flash loader in sram
2016-11-18T13:37:52 ERROR /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/flash_loader.c: flash loader run error
2016-11-18T13:37:52 WARN /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: l1_stlink_flash_loader_run(0x8005480) failed! == -1
2016-11-18T13:37:52 WARN /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c:
write_half_pages failed == -1
  0/  1 pages written
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Starting verification of write complete
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Flash written and verified! jolly good!
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Attempting to write 128 (0x80) bytes to stm32 address: 134239488 (0x8005500)
Flash page at addr: 0x08005500 erased
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Finished erasing 1 pages of 128 (0x80) bytes
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Starting Half page flash write for STM32L core id
2016-11-18T13:37:52 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/flash_loader.c: Successfully loaded flash loader in sram
2016-11-18T13:37:55 ERROR /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/flash_loader.c: flash loader run error
2016-11-18T13:37:55 WARN /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: l1_stlink_flash_loader_run(0x8005500) failed! == -1
2016-11-18T13:37:55 WARN /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c:
write_half_pages failed == -1
  0/  1 pages written
2016-11-18T13:37:55 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Starting verification of write complete
2016-11-18T13:37:55 INFO /Users/tsteinholz/Development/LinkLabs/Public/stlink/src/common.c: Flash written and verified! jolly good!
```

Expected/description:
* no errors / warnings
* flash to take 1-30 seconds, not 10+ minutes

Thank you

tsteinholz, 2016-11-18:
I have also been able to flash to both 0x0 and 0x8000000 over ocd with 
```
openocd -f interface/stlink-v2.cfg -f target/stm32l0.cfg -f ocd-murata.cfg
```

xor-gate, 2016-11-23:
Sorry for not reacting, i'm unsure why this is happening. Make sure your wiring is correct.

xor-gate, 2017-01-14:
Have you got it working faster or did you gave up?

xor-gate, 2017-03-18:
I'm closing this due to inactivity. Feel free to try latest v1.3.1 release or master. Feel free to comment or open a new issue.

closed 2017-03-18 by xor-gate

MikhailNatalenko, 2020-01-20:
I had a very similar issue. GDB was REALY slow, at each step. I added a broken field to the "Expressions" table. (I tried to calculate something odd in an odd way). I didn't realize that the wrong expression could affect that much. I removed this expression and now everything is ok.