stlink-org/stlink #514: Build on Linux Mint

NotOnBoard, 2016-11-21:
Has anyone managed to build stlink on Linux Mint 18.  I have not been able to.  If you have how did you do it please.

xor-gate, 2016-11-21:
Hi @NotOnBoard,

Have you read this document? As mint is debian/ubuntu based it should work out of the box.
https://github.com/texane/stlink/blob/master/doc/compiling.md

NotOnBoard, 2016-11-21:
Thanks for responding.  What do you mean "it should work out of the box", should there be an executable ready to run, or should I be able to build it with make, or build it with something else ?

I have read the Compiling section.  I have installed the prerequisits and tried:
" Run from the root of the source directory:
$ make release  "
but I get "No rule to make target..."

I suspect there are some steps missing in the Compiling instructions, which those more linux savy will know, but I don't.  So any advice would be appreciated.

xor-gate, 2016-11-21:
I mentioned: "it should compile out of the box". Did you do a `git clone` of `https://github.com/texane/stlink` from the latest sources. Else the `release` target is not existing.

NotOnBoard, 2016-11-21:
Oh that's bizarre, why would you edit your post to claim you said something different ?   I still have the original unedited version on the screen in front of me.  Your original post said "it should work out of the box".  That wasn't an issue, I just wanted make sure I understood what you meant so I can get stlink to work.

xor-gate, 2016-11-22:
I understand you confusion, but I re-read my response and saw it missed a essential word.
Did you got it compiling in the mean time or still struggling?

xor-gate, 2016-11-23:
Any progress?

xor-gate, 2016-12-06:
Closing due to no response. This is related to #523 as Mint is based on Ubuntu/Debian.

closed 2016-12-06 by xor-gate

Nightwalker-87, 2020-03-25:
Closed with commit 0c63a2cffc7b7fca817d4c7fb896dbe696341ea3.