stlink-org/stlink #515: Virtual COM Port on stlinkV2-1 support

aguegu, 2016-11-22:
Programmer/board type: Stlink/v2-1

http://www.playembedded.org/blog/en/2015/02/10/hello-chibios

I read this great tutorial on playembedded. It says:

> string “Hello World” is printed on Serial Driver 2 that actually is connected to ST Link V2-1 and is figured in Device Manager as ST Virtual COM port. Connecting to this COM port using a terminal we can capture this string.

As I tried it, I got nothing from /dev/stlinkv2_1 with `minicom`. Well, with modifying the setup to SD1 and an extra USB-USART(cp2102) attached, I got `hello world` with minicom. 

I am not certain whether stlink got this feature covered.

xor-gate, 2016-11-22:
"Virtual com port" is just an emulated serial port. There are two options:

* ARM semihosting (which is currently implemented in st-util)
 * See http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0471m/pge1358787045051.html
* ARM instrumentation macro cell (ITM) -> which is probably is what the ST VCP uses. The firmware writes characters to trace port 0 (ITM port 0).
  * See http://percepio.com/2016/06/09/arm-itm/
  * And http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/BABCCDFD.html

aguegu, 2016-11-23:
@xor-gate great, so how to use it with minicom?

xor-gate, 2016-11-23:
There is no "real serial port" its emulated over the debug channel (usb) and no device node/com port is created by the OS. So you are unable to use it with minicom or whatsoever. The `st-util` tool only currently supports the ARM semihosting "debug channel" and the ITM port is not implemented.

closed 2016-11-23 by xor-gate

aguegu, 2016-11-24:
Thanks, good to know that.