stlink-org/stlink #516: Bluepill 128K support

Testato, 2016-11-25:
the Bluepill board si very famous, it is the cheapest F103 board on the market (only 2$)
It have a 128K flash, this is another advantage, but it is a hidden advantage because, nobody know why, but is there two F103C8 on the market, one is 64KB and the other is 128KB.
The datasheet for C8 version tell that this is a 64KB, but if you try to upload until 128KB it work always, so probably the C8 64KB do not exist.

So may you use this "hack" officially ? 
https://github.com/texane/stlink/issues/305#issuecomment-141833509

xor-gate, 2016-12-01:
Sorry for the late answer, this seems weird. If it is undocumented by ST the stlink project can not support this.

Testato, 2016-12-01:
The stm32duino project, for example, officially support this, by creating two F103C8 board, the user can choose the right one.
Instead of integrate the mentioned hack, maybe there is a way for test the real Flash available on the chip befoure flash it ?
For example the integrated Serial bootloader of the F103C8 mcu correctly show the real Flash available.
If you want i can post the result

xor-gate, 2016-12-01:
I'm not sure where you get this info but this is what I can find http://wiki.stm32duino.com/index.php?title=Blue_Pill.

Testato, 2016-12-01:
yes, i know that the datasheet for C8 version tell 64KB, it is a weird commercial bug :-)

from stm32duino forum (where officially ST enginears working on the official ST arduino core):
http://www.stm32duino.com/viewtopic.php?t=597
> Finally, about the 128kb, that is a nice free upgrade courtesy of ST. All the c8 we have checked so far had 128KB in reality, enjoy ;)

Here there is the board.txt, where you can find the two version implemented for the F103C8, so you can use all 128KB for your project:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/boards.txt#L181

> ## STM32F103C8 -------------------------
> genericSTM32F103C.menu.device_variant.STM32F103C8=STM32F103C8 (20k RAM. 64k Flash)
> genericSTM32F103C.menu.device_variant.STM32F103C8.build.cpu_flags=-DMCU_STM32F103C8
> genericSTM32F103C.menu.device_variant.STM32F103C8.build.ldscript=ld/jtag_c8.ld
> genericSTM32F103C.menu.device_variant.STM32F103C8.upload.maximum_size=65536
> genericSTM32F103C.menu.device_variant.STM32F103C8.upload.ram.maximum_size=20480
> genericSTM32F103C.menu.device_variant.STM32F103C8.upload.flash.maximum_size=65536
> 
> ## STM32F103CB -------------------------
> genericSTM32F103C.menu.device_variant.STM32F103CB=STM32F103CB (20k RAM. 128k Flash)
> genericSTM32F103C.menu.device_variant.STM32F103CB.build.cpu_flags=-DMCU_STM32F103CB
> genericSTM32F103C.menu.device_variant.STM32F103CB.build.ldscript=ld/jtag.ld
> genericSTM32F103C.menu.device_variant.STM32F103CB.upload.maximum_size=131072
> genericSTM32F103C.menu.device_variant.STM32F103CB.upload.ram.maximum_size=20480
> genericSTM32F103C.menu.device_variant.STM32F103CB.upload.flash.maximum_size=131072
>

Testato, 2016-12-01:
And this is the Bootloader message, it clearly tell that the MCU have 128KB.
```
stm32flash 0.4
http://stm32flash.googlecode.com/
Using Parser : Raw BINARY
Interface serial_w32: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (Medium-density)
- RAM        : 20KiB  (512b reserved by bootloader)
- Flash      : 128KiB (sector size: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote address 0x08001bc0 (100.00%) Done.
```

I have personally two of this C8 board, and I can write untill 128KB without problem.

Testato, 2016-12-01:
I asked the stm32duino author to update the Wiki ;-)
http://www.stm32duino.com/viewtopic.php?f=28&t=1598

Testato, 2016-12-01:
Updated Bluepill Wikipage
http://wiki.stm32duino.com/index.php?title=Blue_Pill#128KB_Flash_on_C8_version

Nightwalker-87, 2020-02-26:
Reopening this as some CKS32F103C8T6 chips indeed come with 128 kB flash. A way should be found to deal with this properly.

Nightwalker-87, 2020-03-25:
@Testato: We may implement it in such a way that it is tied to the detection of CKS32F103C8T6 chips which come with 128k, as far as I'm aware of, but this clearly needs some investigation first. I agree with @xor-gate that we must ensure to stay addicted to official specs primarily.

hierophect, 2020-03-26:
@Nightwalker-87 I do not believe that this extra memory is restricted to the CKS32F103C8T6 - I believe I actually saw a confirmation from an ST rep that the true F103C8T6 chips had this extra memory, though I'm having trouble locating the thread now. 

It is also worth noting that the official ST Flash application will happily load up this chip with up to 128k of code :)

hierophect, 2020-03-26:
The temp fix, for anyone who is interested, is to edit the src/common.c file, in the stlink_load_device_params() function, in the following text:

```
} else if (sl->chip_id == STLINK_CHIPID_STM32_L1_CAT2) {
    sl->flash_size = (flash_size & 0xff) * 1024;
} else if ((sl->chip_id & 0xFFF) == STLINK_CHIPID_STM32_L1_HIGH) {
    // 0 is 384k and 1 is 256k
    if ( flash_size == 0 ) {
        sl->flash_size = 384 * 1024;
    } else {
        sl->flash_size = 256 * 1024;
    }
} else {
    sl->flash_size = flash_size * 1024;
}
```
replacing the final else with

```
} else {
     sl->flash_size = flash_size * 1024;
     /* HACK for STM21F103C8T6 */
     if (sl->flash_size == 64 * 1024 && getenv("C8T6HACK"))
       sl->flash_size = 128 * 1024;
}
```

Recompile and install from source, and call with the preface `C8T6HACK= st-flash write etc`

Nightwalker-87, 2020-03-27:
Related to #305.

Nightwalker-87, 2020-03-31:
@hierophect: Surprise - Digging through the whole project revealed some interesting features, which seem to have been kicked into the long grass - without any documentation *sigh*. However I think this is what you have been looking for. Sry to see that you waited for nothing... :-(

closed 2020-03-31 by Nightwalker-87