stlink-org/stlink #517: Yet another "unknown chip id"

spe-ciellt, 2016-11-27:
- Programmer/board type:Stlink/v2-onboard
- Programmer firmware version: STSW-LINK007 2.28.16
- Operating system: e.g Linux
- Stlink tools version: git 3de5cf05619926e09e885f8980478a0c128a0f08
- Target chip (and optional board):  STM32L152RE (STM32L152 Nucleo)

Sorry for this, feeling really stupid about this. Only reference I found is #477, which is closed as duplicate. But I cannot find the duplicate.

I have a nucleo board L152RE that worked under stlink. All of a sudden it returns "unknown chip id 0".  After upgrading to V2J28M16 using Java tool under Linux it now returns "unknown chip id 0xa05f0000". But (as in #477 mentioned above) it works while holding down the reset button.
./st-info --probe
Found 1 stlink programmers
./st-info --chipid (without pressing the reset button)
(no answer)
./st-info --chipid (when pressing the reset button)
0x0437

Thanks!
/Stefan

xor-gate, 2016-11-27:
You are right, but there is something weird happening. It sometimes works, sometimes not (tested on a stm32l152rct6 discovery board):

```
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --chipid
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --probe
Found 1 stlink programmers
 serial: 503f7206506752553329033f
openocd: "\x50\x3f\x72\x06\x50\x67\x52\x55\x33\x29\x03\x3f"
  flash: 262144 (pagesize: 256)
   sram: 32768
 chipid: 0x0427
  descr: L1 Medium-Plus-density device
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --chipid
0x0427
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --probe
Found 1 stlink programmers
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --chipid
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --chipid
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --chipid
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --probe
Found 1 stlink programmers
Jerrys-MBP-405:stlink jerry$ ./build/Release/st-info --probe
Found 1 stlink programmers
 serial: 503f7206506752553329033f
openocd: "\x50\x3f\x72\x06\x50\x67\x52\x55\x33\x29\x03\x3f"
  flash: 262144 (pagesize: 256)
   sram: 32768
 chipid: 0x0427
  descr: L1 Medium-Plus-density device
```

spe-ciellt, 2016-11-27:
I connected my Black Magic Probe to SWDIO and SWCLK of the L152 and it seems that it also only identifies the MCU if the MCU is in reset (monitor swdp_scan).
I have not connected reset since my STM32F1x target (chinese cheap board) works without reset, so I figured it isn't needed. Maybe I am wrong?

xor-gate, 2016-12-01:
The reset pin is necessary to make sure the MCU is in the correct state. The stlink tools use this mechanism.

spe-ciellt, 2016-12-02:
The software that I managed to download was inferior somehow. After a some hacking it is back to working.

I got it back by using a Black Magic Probe, but probably any external programmer would work.
By removing the jumpers for ST-LINK/NUCLEO and connect the SWDIO and SWCLK of the programmer there I managed to talk to the chip (don't forget the +3.3V and GND). But only when it was in reset (pushing the the button).

So I connected the reset of the programmer to the T_NRST signal of the board. Starting GDB and connecting to my Black Magic Probe I configured it to "configure connect under reset" (monitor connects_srst enable). Then running "monitor swdp_scan" worked second time, I attached to the target and downloaded the image that I knew wasn't broken.
Disconnected everything, put the jumpers back in place and, voila, st-util -m found the target.

So somewhere in there maybe someone can learn something, that is my motivation for this long ending blurb.

closed 2016-12-02 by spe-ciellt