stlink-org/stlink #519: Compile under Ubuntu

gorynch, 2016-11-29:
Hi
I'm trying to compile stlink under Ubuntu 15.04
I have C maker, gcc, libusb, libusb-dev.
Type in Terminal
`$ git clone git://github.com/texane/stlink.git`
and 

> 
> $ make release
> [RELEASE]
> Makefile:27: recipe for target 'release' failed
> make: *** [release] Error 2

There are a lot of manuals how to install stlink, BUT all of them using **autogen.sh**. In my case there is no such file.
Any idea?

xor-gate, 2016-11-29:
Try running `make clean && make release` this probably due to missing `build/Release` folder and CMake files. Did you git cloned the latest version from `master` branch?

vijayandra, 2016-11-29:
When STLINK connected with Ubuntu, softreset from firmware does not work. Debugged very long to find issue but now seems Linux/Qstlink combination is issue.

julienrhodes, 2017-01-09:
I had to run `cmake .` to make the build files prior to make.

xor-gate, 2017-01-09:
Then you are building **inside the source folder**, I recommend:

```
mkdir build && cd build && cmake .. && make
```

xor-gate, 2017-01-14:
I'm closing this, as it is inside the [doc/compiling.md](https://github.com/texane/stlink/blob/master/doc/compiling.md) documentation.

closed 2017-01-14 by xor-gate