stlink-org/stlink #520: Linux mint installation issue, With latest git code issue in code:debuild -uc -us

vallab, 2016-12-01:
I have followed all the steps in https://github.com/texane/stlink/blob/master/doc/compiling.md
installed all the required dependencies.
please guide me how to compile 
it fails after debuild -uc -us
Terminal output:
dpkg-buildpackage -rfakeroot -D -us -uc
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: source package libstlink
dpkg-buildpackage: source version 1.2.1
dpkg-buildpackage: source distribution unstable
dpkg-buildpackage: source changed by Andrew 'Necromant' Andrianov <andrew@ncrmnt.org>
 dpkg-source --before-build stlink
dpkg-buildpackage: host architecture amd64
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9)
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -D -us -uc failed

**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2016-12-01:
Hi @vallab I have edited your content of the issue. If you read the error message correctly you need to install debhelper (https://packages.debian.org/jessie/debhelper). 

It should be added to the documentation (for debian based distros).

vallab, 2016-12-01:
Thanks xor-gate it compiled

xor-gate, 2016-12-01:
Great, I need to add this to the documentation before v1.3.0 (reopening so I dont forget).

commit 2cb33d45 (2017-01-14): Add note about debhelper for compiling the debian package. Fixes #520
closed 2017-01-14 by xor-gate