stlink-org/stlink #523: compilation on ubuntu 16.10

yosoufe, 2016-12-04:
Hi
I have recently installed ubuntu 16.10 and I am trying to install stlink driver. Unfortunately I cannot compile the code. What am I doing wrong? problems with ubuntu 16.10?

[CMakeFiles_output.zip](https://github.com/texane/stlink/files/629692/CMakeFiles_output.zip)

This is the output of the **make release** command:

# make release

-- The C compiler identification is GNU 6.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at cmake/Version.cmake:25 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  CMakeLists.txt:15 (include)

-- Performing Test C_SUPPORTS_STD_GNU99
-- Performing Test C_SUPPORTS_STD_GNU99 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Found Libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.20.9
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Manpage generation disabled
-- Configuring incomplete, errors occurred!
See also ".../Toolchain/stlink/build/Release/CMakeFiles/CMakeOutput.log".
See also ".../Toolchain/stlink/build/Release/CMakeFiles/CMakeError.log".
Makefile:42: recipe for target 'build/Release' failed
make: *** [build/Release] Error 1

# **And this is the CMakeError.log:**

Performing C SOURCE FILE Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS failed with the following output:
Change Dir: .../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_f6585/fast"
make[1]: Entering directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_f6585.dir/build.make CMakeFiles/cmTC_f6585.dir/build
make[2]: Entering directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f6585.dir/src.c.o
/usr/bin/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS   -Wmissing-variable-declarations -o CMakeFiles/cmTC_f6585.dir/src.c.o   -c .../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-Wmissing-variable-declarations'; did you mean '-Wmissing-declarations'?
CMakeFiles/cmTC_f6585.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_f6585.dir/src.c.o' failed
make[2]: *** [CMakeFiles/cmTC_f6585.dir/src.c.o] Error 1
make[2]: Leaving directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_f6585/fast' failed
make[1]: *** [cmTC_f6585/fast] Error 2
make[1]: Leaving directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
Performing C SOURCE FILE Test C_SUPPORTS_WSHORTEN_64_TO_32 failed with the following output:
Change Dir: .../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_04dd3/fast"
make[1]: Entering directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_04dd3.dir/build.make CMakeFiles/cmTC_04dd3.dir/build
make[2]: Entering directory '...Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_04dd3.dir/src.c.o
/usr/bin/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WSHORTEN_64_TO_32   -Wshorten-64-to-32 -o CMakeFiles/cmTC_04dd3.dir/src.c.o   -c .../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-Wshorten-64-to-32'
CMakeFiles/cmTC_04dd3.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_04dd3.dir/src.c.o' failed
make[2]: *** [CMakeFiles/cmTC_04dd3.dir/src.c.o] Error 1
make[2]: Leaving directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_04dd3/fast' failed
make[1]: *** [cmTC_04dd3/fast] Error 2
make[1]: Leaving directory '.../Toolchain/stlink/build/Release/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }

xor-gate, 2016-12-04:
It seems (your) cmake (version) chokes on this: https://github.com/texane/stlink/blob/master/cmake/Version.cmake#L25-L26

Which version are you running `cmake --version`?

yosoufe, 2016-12-04:
it is cmake version 3.5.2
How can I solve it? I am newbie in these topics

xor-gate, 2016-12-05:
I need to investigate this.

yosoufe, 2016-12-05:
I downloaded the zip file

xor-gate, 2016-12-05:
I have proposed a fix. It was due to developing and building from a git tree (and version detection from git), and I have not checked against zipfile source building.

Thanks for reporting, it should work now!

xor-gate, 2016-12-05:
I need to fix this (also) on older git versions, reopening...

newtoniant, 2017-01-19:
I too am getting this build error on Ubuntu 16.04 LTS, cmake version 3.5.1

Errors in the log:
unrecognized command line option '-Wmissing-variable-declarations'
unrecognized command line option '-Wshorten-64-to-32'

xor-gate, 2017-01-25:
@newtoniant this should not be relevant when using a newer zipfile (master.zip).

xor-gate, 2017-01-28:
Closing, preparing for v1.3.0. Feel free to open a new issue after the new release has happend.

closed 2017-01-28 by xor-gate

alukichev, 2017-04-06:
This appears also when building from git tree with `git checkout 1.3.1; make debug` on Fedora 25 (gcc 6.3.1, cmake 3.6.2). [CMakeOutput.log](https://github.com/texane/stlink/files/902427/CMakeOutput.log.txt) and [CMakeError.log](https://github.com/texane/stlink/files/902426/CMakeError.log.txt) are attached.

In particular, errors are the same:

> Performing C SOURCE FILE Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS failed with the following output:
...
> /usr/lib64/ccache/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS   -Wmissing-variable-declarations -o CMakeFiles/cmTC_df367.dir/src.c.o   -c /home/blablabla/stlink/build/Release/CMakeFiles/CMakeTmp/src.c
> cc: error: unrecognized command line option '-Wmissing-variable-declarations'; did you mean '-Wmissing-declarations'?

...
> Performing C SOURCE FILE Test C_SUPPORTS_WSHORTEN_64_TO_32 failed with the following output:
...
> /usr/lib64/ccache/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WSHORTEN_64_TO_32   -Wshorten-64-to-32 -o CMakeFiles/cmTC_0ceee.dir/src.c.o   -c /home/blablabla/stlink/build/Release/CMakeFiles/CMakeTmp/src.c
> cc: error: unrecognized command line option '-Wshorten-64-to-32'

 Requesting to reopen the issue.

alukichev, 2017-04-06:
Problem solved. I was missing libusb headers : - ) . Sorry to bother.

fedorg, 2018-05-05:
The easiest solution is to remove the `.git` directory and try again:
`rm -r build && sudo rm -rf .git && make`
It works on Linux Mint 18.3 and probably everywhere.

commit 0c63a2cf (2016-12-05): cmake/Version.cmake: Fix version detection when building from zipfile (non-git tree). Fixes #523