stlink-org/stlink #524: Build Problems on Linux Mint 18

upost, 2016-12-07:
I am trying to build on Linux Mint 18. Prerequisites are installed.

Using git clone, it does not work at all. make says it has no targets.

When unpacking the zip, make does work.
Then I do cd build/Release and sudo make install. This also works.
Among other lines, it says: -- Installing: /usr/local/lib/libstlink.so.1
When running st-info (or any of the other binaries):
st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory
As stated before, the library is there and is readable.

I never had problems with your project like this before I switched from Debian to Linux Mint, but I have no idea what the cause is. Then I saw a (now closed) issue here also concerning building on Mint. So maybe there's indeed something wrong not only on my PC. Any hint?

Thanks in advance

xor-gate, 2016-12-07:
You shared library cache is not updated (same as issue https://github.com/texane/stlink/issues/478).
I have added a note about this but it could be improved:
https://github.com/texane/stlink/commit/be66bbf200c718904514b044ba84d64a36456218

xor-gate, 2017-01-14:
Thanks for your report! Closing this as I have added it to the documentation. Have fun.

closed 2017-01-14 by xor-gate