stlink-org/stlink #529: Compile under Ubuntu 16.04

df8508, 2016-12-15:
I use Ubuntu 16.04
I am new to linux and i am starting to work on it recently, 
I have my stlink/v2 work on windows. but it not works on linux.
My problem is following:

i can not find the 
./autogen.sh
./configure inside the directory .
i follow this 
# Compile under Ubuntu #519 
but it cannot help, 
but my commend 'make' works
however, due to some webpage , the st-util, st-flash and st-info are locate in under the main directory but located in different one, is it correct ?
thanks 

[image]
 

**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [X] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: STSW-LINK007 2.27.15
- [X] Operating system: Linux 
- [X] Stlink tools version and/or git commit hash: v1.2.0-155/gecc-31f7
- [X] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`

xor-gate, 2016-12-15:
It works as expected, problem due some missing documentation or read incorrectly:

* Where the final executables are placed
* How to install the build

Have you read this document (which is linked from the mainpage README.md)?:
https://github.com/texane/stlink/blob/master/doc/compiling.md#build-from-sources

At the end of the section there is a note how to install into `$HOME` or root `/`

xor-gate, 2016-12-30:
Did you managed to find and get the executables running?

df8508, 2016-12-31:
> Did you managed to find and get the executables running?

Yes.thanks you a lot ~

xor-gate, 2017-01-01:
Thank you for your feedback. Have fun!

closed 2017-01-01 by xor-gate