stlink-org/stlink #530: Add new command to gdb: "reset"

SL-RU, 2016-12-19:
Sometimes it's needed to often reset your MCU, but pressing reset button may be rather annoying, and doing every time kill&run in gdb annoying tooo. Maybe it's possible to add new command to gdb server? Something like "reset", like "flash" and etc? That'll be awesome! 

What do you think?

Thank you,
The stlink project maintainers

xor-gate, 2016-12-30:
You should get some knowledge about GDB batch scripting which can solve this. 
See https://sourceware.org/gdb/download/onlinedocs/gdb/index.html

You should run something like:

`gdb --batch --command=myscript.gdb`:

**myscript.gdb**:

```
target remote localhost:1234
monitor reset halt
file app.elf
load app.elf
monitor reset run
```

xor-gate, 2017-03-01:
Closing due to inactivity. Feel free to comment if you still have problems.

closed 2017-03-01 by xor-gate