stlink-org/stlink #532: Board won't start up after flashing

Joe-Merten, 2016-12-25:
Compiled most recent sources
- ecc31f764d00447acbc0198c65a0f79d47e7e365
- Host = Kubuntu 16.04 64 bit
- `st-flash --version` = v1.2.0-155-gecc31f7

Stm target boards:
- Stm32F401C-Disco (chip-id = 0x0423, `st-info --descr` = `F4 device (low power)`)
- Stm32F411E-Disco (chip-id = 0x0431, `st-info --descr` = `F4 device (low power) - stm32f411re`)
- both Stlink/v2-onboard

Stlink commandline:

    build/Release/st-flash --reset write "myprog.bin" 0x8000000

Output:

```
2016-12-25T20:47:55 INFO src/common.c: Loading device parameters....
2016-12-25T20:47:55 INFO src/common.c: Device connected is: F4 device (low power), id 0x6423
2016-12-25T20:47:55 INFO src/common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 16384 bytes
2016-12-25T20:47:55 INFO src/common.c: Attempting to write 209516 (0x3326c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08020000 erasedEraseFlash - Sector:0x5 Size:0x20000 
2016-12-25T20:47:58 INFO src/common.c: Finished erasing 6 pages of 131072 (0x20000) bytes
2016-12-25T20:47:58 INFO src/common.c: Starting Flash write for F2/F4/L4
2016-12-25T20:47:58 INFO src/flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 12908
2016-12-25T20:48:03 INFO src/common.c: Starting verification of write complete
2016-12-25T20:48:08 INFO src/common.c: Flash written and verified! jolly good!
```

Expected/description:

After finished writing to flash, the flashed binary don't start up (regardless if I call `st-flash` with or without `-reset` parameter). I need to press the "RESET" button of the discovery board to start my application.

What's the way to automatically start the flashed binary?

xor-gate, 2016-12-25:
Thats actually a very good question. Probably we need a option like `run-after-flash`. I'm not 100% sure why it doesn't reset after flash currently.

Joe-Merten, 2016-12-25:
Looking into `src/tools/flash.c`, I saw some code which is executed when I pass the `--reset` option:

    if (o.reset) {
        stlink_jtag_reset(sl,2);
        stlink_reset(sl);
    }

but this seems not to reset the microcontroller.

Joe-Merten, 2016-12-25:
Trying the `reset` command also doesn't start my flashed application:

```
$ build/Release/st-flash reset
st-flash 1.2.0-155-gecc31f7
2016-12-26T00:54:15 INFO src/common.c: Loading device parameters....
2016-12-26T00:54:15 INFO src/common.c: Device connected is: F4 device (low power), id 0x6423
2016-12-26T00:54:15 INFO src/common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 16384 bytes
```

Joe-Merten, 2016-12-26:
Added `stlink_run(sl);` after `stlink_reset(sl);` won't help.

xor-gate, 2016-12-30:
I have just merged a possible fix from PR #534. Could you test latest master?

Joe-Merten, 2017-01-04:
Testet with most recent 01b9e1682104386a0dc092f9f2481a13d599ca93 but same behaviour.

aoreh, 2017-03-13:
Hi! Any progress on this issue?
Have the same behavior :(, with attached and detached SWRST pin. But it looks that device start for a few milliseconds and then holds. It looks that this "starting" happens during call to "stlink_exit_debug_mode"
After manual resetting device works fine.
At the same time if connect device to Win system and program it - it starts working w/o any problem.

xor-gate, 2017-03-14:
SWRST pin is independed of CPU core state (running,halted). Which register-controlled. There is no progress on this issue and I'm happy to receive a correct fix for this.

aoreh, 2017-03-14:
Ok, but why it doesn't work without SWRST pin? Program runs for a few milliseconds (I see log messages from board via UART) and then holds :( As I said - no problem with STM's original SW.

About SWRST: unfortunately work with device remotely, so can't check what happened with oscilloscope :(, may be something wrong with this pin it self.

aoreh, 2017-03-15:
Tried with OpenOCD, didn't re-flash the device, just "reset run" - and device started working.

Nightwalker-87, 2020-03-21:
@JoeMerten @aoreh: I'm currently reviewing old leftovers. What is the current state of this? Can you confirm this this not a hardware-related issue? Have you tried with Release v1.3.0?

Ant-ON, 2021-03-24:
This issue most like fixed by #1071 (software reset rework)

Nightwalker-87, 2021-03-24:
Closed by #1071.

closed 2021-03-24 by Nightwalker-87