stlink-org/stlink #533: st-flash reset can be a hardware reset?

feathertw, 2016-12-27:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [`X`] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [`X`] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.
------------------------------------
I found the command "st-flash reset" can send a reset info to a stm32 chip
but it's not whole system reset,
for example:
If one led is on, but after "st-flash reset", it will still on but I think it should off after boot whole system.

I think it's because "st-flash reset" is "core reset", not "system reset" in st windows stlink tool,
so it will operate arm cpu to reset, the peripheral will not reset,

hope it will be more option let user can do hardware reset =)
------------------------------------------

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2016-12-28:
Thats a good question, there indeed two different resets. There are some other issues related to this. I'm not sure how we can fix this.

xor-gate, 2016-12-30:
I have just merged a possible fix from PR #534. Could you test latest master?

feathertw, 2017-01-01:
Haha, it works now,
And I can find that at Reset Pin of SWD in stm32f4-discovery,
there is a low signal when I key "st-flash reset".

Thank you, and HAPPY NEW YEAR =)

FEA

xor-gate, 2017-01-01:
Thanks for reporting back! 
Kind regards,

feathertw, 2017-01-09:
Hello again =)

I found that a strange things in st-link reset,
I use the stlink V2 (the image I post, but actually I am not sure it's from
st or not)
I can use only GND,SWDIO,SWCLK(without reset pin) to burn code into stm32
And I can use ST-Link Utility in windows "System Reset" to reset whole
stm32 system(including reset peripheral)

I have no idea why I don't connect reset pin and it still can work hardware
reset,
But in texane stlink, it need connect reset pin to operate hardware reset,
Is it possible to do hardware reset without connecting reset pin in texane
stlink?
if it can, it's more flexible and can save the reset pin connecting

Thank you =)
Fea

[image: 內置圖片 1]

​

xor-gate, 2017-01-09:
Hi @feathertw could you open a new issue and note the system reset register write instead of asserting the reset line? As system reset command is unrelated to hardware reset. You replied to this issue by email and the image is not visible because you need to edit the issue at github itself.

Thanks!

rewolff, 2020-03-17:
Hi Nightwalker, 
I've run into this "semi-hardware reset" through SWD issue myself. Do you remember another issue related to this from me? I'm pretty sure I have reported it, but just like here it might be hidden as a comment to another issue. 
Anyway. I'll try to test the ASAP, which worst case is 2 weeks from now (by which time I'll have forgotten).....

Nightwalker-87, 2020-03-17:
I'll have a look and let you know. Please do not recycle this issue in the meanwhile.

Nightwalker-87, 2020-03-17:
Please check https://github.com/texane/stlink/issues?q=+label%3Aerror%2Freset+. As you can see most of these topics are open or have been reopenend due to their unclear state. Reset-issues popped up several times but were never really addressed. I feel like we should go back to start with this topic of reset issues. There is only one PR i can find which (and also only partially addresses this group of issues) and so hardly any contributions made it into the code.

rewolff, 2020-03-17:
FYI: I prevented my colleague from going to work, so now I'm alone at work. 
I have a board that multiplexes three seven segment displays. So, "CPU halted or crashed" is distinguishable (one segment lit up with normal number)  from "CPU reset" (nothing lit). 

Both the older version as the current git one are now unable to program my board when it had been programmed before. the first time I see the CPU getting halted and then erasing, "flash loader run error". 

IIRC, I updated my git stlink sources a few weeks ago. I then tested it and was satisfied it was usable. Now I can't get it to work. Either both  the modules I picked out of the box were broken that they did do the display enough to show the normal startup when powered through my STLINK, but then failed to program. Or I checked out the wrong branch on my sources. Or stlink doesn't work, neither the two weeks ago version nor the one today. I don't have time to investigate further right now. But no "fixed! " confirmation from me at this point in time.

Nightwalker-87, 2020-03-17:
If possible I would suggest the following:
1) Clean your build system (ensure there are no leftovers), checkout Release v1.6.0 and then do a clean install from source. With this we can likely rule out that there are any local leftovers that may cause any corruption or strange behaviours.
2) Open a new ticket (even if there may be several related ones) and use the available issue template to describe _your specific_ system and hardware.
3) Rediscuss

Currently it remains difficult to maintain and keep track of all various reset-related issues. This wants cleaning up first, which is on the project side. I'm currently trying to clean up with this.

Nightwalker-87, 2020-03-21:
It reads, as if this has been fixed in PR #534 already.
@rewolff: You didn't use v1.3.0, did you?

rewolff, 2020-03-22:
No. On my current system with the "build system change" between 1.2.0 and 1.3.0  something was introduced that causes the build to fail on my system. This was only fixed between 1.5.0 and 1.6.0 . 

So I can't test 1.3.0 to 1.5.0 for you. Sorry.

Nightwalker-87, 2020-03-23:
@rewolff: That's ok, don't bother about it too much. Your feedback is helpful as well.
I think we may have false friends here looking at the link to the PR. I'll dereference this.

Nightwalker-87, 2021-03-20:
Closed by #1107.

closed 2021-03-20 by Nightwalker-87