stlink-org/stlink #537: add support for STM32F412

Androude, 2017-01-02:
- [X] Programmer/board type: Stlink/v2-onboard
- [X] Operating system: Linux Ubuntu 16.04
- [X] Stlink tools version and/or git commit hash: st-util 1.2.0-155-gecc31f7-dirty
- [X] Stlink commandline tool name: `st-flash`, `st-util`
- [X] Target chip (and optional board): 32F412 (32F412GDISCOVERY)

The STM32F412 micro-controller line is not supported. 

Output:

st-util 1.2.0-155-gecc31f7-dirty
2017-01-02T18:57:39 INFO /home/androude/apps/stlink/src/stlink/src/common.c: Loading device parameters....
2017-01-02T18:57:39 WARN /home/androude/apps/stlink/src/stlink/src/common.c: unknown chip id! 0x30006441

Expected/description:

I have fixed the issue locally by adding appropriate chipid-params structure and if-condition to flash loader. Using the 32F412GDISCOVERY device I successfully entered debug mode with gdb from eclipse. 

I have never made a pull-request so fixing this issue could be the first one for me.

xor-gate, 2017-01-02:
Best way is to fork the project, create a feature branch and the send a pull request:
https://help.github.com/articles/proposing-changes-to-your-work-with-pull-requests/

xor-gate, 2017-01-02:
Thanks for your contribution with #538 !

closed 2020-03-17 by Nightwalker-87