stlink-org/stlink #542: gdb/gdb/inferior.c:361: internal-error: find_inferior_pid: Assertion `pid != 0' failed.

df8508, 2017-01-12:
hello, i make the makefile myself to out a elf and run the elf following the read me.
it can steps after the first run, 
however once I kill it and run again, 
it appears the problem 
gdb/gdb/inferior.c:361: internal-error: find_inferior_pid: Assertion `pid != 0' failed.
as I mentioned.

i search through google but is it a gdb error?
my gdb using  is arm-none-eabi-gdb --version
GNU gdb (7.10-1ubuntu3+9) 7.10
I appreciate if you can help me~
 

[image]
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2017-01-12:
You should take your time to fill in the template which popups when you created a new issue. Please reorganise your issue comment.

It seems to be a bug in gdb itself and not in `st-util`.

xor-gate, 2017-01-14:
I will close this, as it is unrelated to the project. But thanks for reporting and asking!

closed 2017-01-14 by xor-gate