stlink-org/stlink #543: stm32vldiscovery board no longer writable with st-flash

jmfriedt, 2017-01-14:
I used to flash the STM32VLdiscovery [1] with st-flash. The curent git version of st-flash will no longer parse the command line as given in the documentation (doc/tutorial.md):
$ ./st-flash write v1 in.bin 0x8000000
invalid command line
I trace the issue to src/tools/flash_opts.c
            if(o->format == FLASH_FORMAT_BINARY) {    // expect filename and addr
                if (ac != 2) return -1;
in the case of stlink v1, we have 3 args, not 2 (v1 file.bin address), hence the failure to parse the command line I believe.

[1] http://www.st.com/en/evaluation-tools/stm32vldiscovery.html

- [X] Programmer/board type: Stlink/v1
- [X] Operating system: e.g Linux
- [X] Stlink tools version and/or git commit hash: v1.2.0/git-c0106c53 
- [X] Stlink commandline tool name: st-flash
- [X] Target chip: STM32VLdiscovery -- STM32F100

xor-gate, 2017-01-14:
Thats a good catch, the tutorial.md is actually an conversion from an very old pdf document. It has not been read/updated and reviewed for some time. It is indeed incorrect. Probably we should change it to the current commandline arguments.

The project got some time ago man-pages which are converted from markdown to man which er under the `doc/man` folder.

Are you able to use the tool with the "new" arguments on your board or is it utterly broken?

Thank you for your report!

jmfriedt, 2017-01-16:
I confirm that the issue is only with the documentation and that st-flash can write properly in a STM32VLDicovery board by using the same command line as given in the documentation, without the v1 option:
st-flash write file.bin 0x8000000

xor-gate, 2017-01-16:
Thats good news, we should update the documentation then. With the upcoming v1.3.0 release you can also write hex files with st-flash!

commit e05ad172 (2017-01-28): Update doc/tutorial.md with latest commandline arguments. Fixes #543.
closed 2017-01-28 by xor-gate