stlink-org/stlink #545: STM32F103 won't flash

bolorkhuu, 2017-01-25:
Hello,

i'm trying to flash an STM32F103 Board via SWD with an STM32F407 Discovery Board. Unfortunately  it does not work only with STM32F103 Board.

- [ ] Programmer/board type: Stlink/v2
- [ ] Operating system: Mac OS X
- [ ] Stlink tools version stlink: stable 1.2.0 (bottled), HEAD, Poured from bottle on 2015-10-04
- [ ] Stlink commandline tool name: `st-flash`
- [ ] Target chip (and optional board): e.g STM32F103C

Output:

```
st-flash write prog.bin 0x8000000
2017-01-24T18:44:14 INFO src/stlink-common.c: Loading device parameters....
2017-01-24T18:44:14 INFO src/stlink-common.c: Device connected is: F1 High-density device, id 0x10036414
2017-01-24T18:44:14 INFO src/stlink-common.c: SRAM size: 0x10000 bytes (64 KiB), Flash: 0 bytes (0 KiB) in pages of 2048 bytes
```
As you can see, ```  Flash: 0 bytes (0 KiB) in pages of 2048 bytes  ``` 

Does anyone know how to solve it. 
Thank you in advance.

bolorkhuu, 2017-01-25:
**Solved** with ST Link Utility.

the board was _write protected_

xor-gate, 2017-01-25:
Thanks for your report, I think I will add this to the docs so other people will not fail with this.

closed 2017-01-28 by xor-gate

rodrigopex, 2017-03-13:
How can I disable the write protection? Can you help out?

xor-gate, 2017-03-13:
You probably could mass erase first and I thought write protection is cleared then. Or with some openocd commands (it has more features then texane/stlink).

rodrigopex, 2017-03-13:
Thank you for your rapid answer. I've tried erasing the flash, it worked. However, the issue remains. I will try to use some windows computer to see what happen. :\

xor-gate, 2017-03-13:
Have you seen this in the README? 
https://github.com/texane/stlink/commit/0f9e4990ba0cdc295875b219b354fdf42f30a9aa

When using OpenOCD you could probably invoke `monitor stm32f1x unlock 0` commands (untested). See http://openocd.org/doc/html/Flash-Commands.html.

rodrigopex, 2017-03-13:
I've run the ST-LINK Utility (on windows) and I went to Target > Option Bytes... > Disable Read Out Protection (combo box). Done. I am testing the st-util again to see if it solved the problem.

rodrigopex, 2017-03-16:
I could find clue on disabling the Read Out Protection using openOCD following this link: http://www.openstm32.org/forumthread562. I am using the Blue Pill Stm32F103c8t6 a Chinese board. It is just necessary to add the "monitor flash protect 0 0 11 off" line to the init script of openOCD. [Untested]

jeroavf, 2018-10-15:
Hi , I had success unlocking stm32 bluepills using the following commands on openocd terminal session: 
> reset halt 
> flash banks 
> stm32f1x mass_erase 0 
> stm32f1x unlock 0 

After that, the device was ready to be programmed again.

Leonidas-from-XIV, 2019-04-22:
@jeroavf Thanks! This worked for me and `--probe` now returns `65536` (looks like I got one of the few 64kb models) as well as flashing correctly.

Fun bit, when doing the `mass_erase` step it even thinks it is 128kb, but nevermind.
```
STM32 flash size failed, probe inaccurate - assuming 128k flash
flash size = 128kbytes
```

geekshow, 2020-02-22:
Thanks, I was able to unbrick some STM32F437 with a full chip erase with the windows ST-Link utility, now this is working again on linux st-link :-)

commit 0f9e4990 (2017-01-28): README.md: Add note about write protected flash. Fixes #545