stlink-org/stlink #547: Build error: cannot find -lstlink

thehans, 2017-01-30:
- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: unknown
- [X] Operating system: Linux Mint 17.3
- [X] Stlink tools version and/or git commit hash: eea16d762979fcfc28937c784c480c8346247a3a
- [X] Stlink commandline tool name: n/a
- [X] Target chip (and optional board): e.g STM32F103

When attempting to compile from source, the build exits with error "cannot find -lstlink"

Attached is the output of running ```make release``` from the top directory after a fresh git clone
[build.txt](https://github.com/texane/stlink/files/737960/build.txt)

xor-gate, 2017-01-30:
Nice catch, I'm not 100% sure why this happens but on Debian 8 it works (or worked) fine. I will fix this in v1.3.1.

xor-gate, 2017-01-30:
Related to issue #551 and probably fixed with PR #552.

thehans, 2017-01-31:
works now, thanks.

closed 2017-01-31 by thehans