stlink-org/stlink #548: INFO messages print absolute path

cdwilson, 2017-01-30:
(this may border on anal retentiveness...but just wanted to point it out as an FYI)

When using the 1.3.0 release on macOS downloaded from https://github.com/texane/stlink/releases/tag/1.3.0 the `INFO` debug messages print the absolute path from your build machine, i.e.:

`/Users/jerry/Downloads/stlink-master/...`

Here's the full log:

```
$ st-flash write ~/Projects/STM32/hello-nucleo/Debug/hello-nucleo.bin 0x08000000
st-flash 1.3.0
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Loading device parameters....
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Device connected is: F4 device (low power) - stm32f411re, id 0x10006431
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: SRAM size: 0x20000 bytes (128 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Attempting to write 4828 (0x12dc) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Starting Flash write for F2/F4/L4
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 4828
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Starting verification of write complete
2017-01-29T19:59:18 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Flash written and verified! jolly good!
```

If there is an easy way to truncate these in the Mac/Win release packages, it would make for less visual noise in the console output (i.e. change `/Users/jerry/Downloads/stlink-master/src/common.c` --> `src/common.c`)

xor-gate, 2017-01-30:
I agree with you. I will fix this with v1.3.1.

xor-gate, 2017-02-14:
Fixed by https://github.com/texane/stlink/commit/2c0ab7f0eb6cfda5cfbdc08bb9f6760d27c2b667

closed 2017-02-14 by xor-gate

xor-gate, 2017-03-04:
For those who interested I have extracted this trick into my cmake examples repo: 
https://github.com/xor-gate/cmake-by-example/tree/master/filepath

commit 2c0ab7f0 (2017-02-14): logging: Don't print absolute path to source file. Fixes #548