stlink-org/stlink #551: GUI could not be built with cmake due to wrong order of actions (missing dependencies?)

blacklion, 2017-01-30:
I'm trying to update FreeBSD port to post-1.3.0 version, commit 4f460ec. It works flawlessly without gtk-3.0 support, but with gtk-3 installed it try to build GUI and fails, because it try to link stlink-gui BEFORE it links libstlink (both static & dynamic).

I'm attaching two log files: one when gtk-3 is not installed ([stlink-nogiu.txt](https://github.com/texane/stlink/files/740310/stlink-nogiu.txt)), it is successful build. And one with gtk-3 installed ([stlink-gui.txt](https://github.com/texane/stlink/files/740311/stlink-gui.txt)), which fails.

I'm using 

$ cmake --version
cmake version 3.7.2
$

BTW, is it possible to disable stlink-gui build in presence of gtk-3 in system? autoconf-based system allowed this.

xor-gate, 2017-01-30:
Yeah I have broken de build for this. Somebody already fixed it for Arch (and probably also for FreeBSD) with this PR https://github.com/texane/stlink/pull/552.

xor-gate, 2017-02-12:
Ping @blacklion, could you try latest master?

blacklion, 2017-02-12:
Oh, I'm sorry, it works! Latest version are in FreeBSD ports now, works!
Thank you!

closed 2020-03-17 by Nightwalker-87