stlink-org/stlink #553: Error compiling from source

bmetman, 2017-02-01:
- [X] Operating system: Ubuntu 16.04
- [X] Stlink tools version and/or git commit hash: v1.3.0/git-f62895b

I cloned the git repository and tried to build from source using the command 'make release'. However the compilation failed. 

Command line output:

-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- stlink version: 1.3.0
--           Major 1 Minor 3 Patch 0
-- Performing Test C_SUPPORTS_STD_GNU99
-- Performing Test C_SUPPORTS_STD_GNU99 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Could NOT find Libusb (missing:  LIBUSB_LIBRARY LIBUSB_INCLUDE_DIR) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtk+-3.0'
--   No package 'gtk+-3.0' found
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Manpage generation disabled
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master
   used as include directory in directory /home/user/Downloads/stlink-master/src/gdbserver
   used as include directory in directory /home/user/Downloads/stlink-master/src/gdbserver
[... 515 lines cut ...]
make[1]: Leaving directory '/home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
Performing C SOURCE FILE Test C_SUPPORTS_WSHORTEN_64_TO_32 failed with the following output:
Change Dir: /home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_42dc8/fast"
make[1]: Entering directory '/home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_42dc8.dir/build.make CMakeFiles/cmTC_42dc8.dir/build
make[2]: Entering directory '/home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_42dc8.dir/src.c.o
/usr/bin/cc    -std=gnu99 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -DC_SUPPORTS_WSHORTEN_64_TO_32   -Wshorten-64-to-32 -o CMakeFiles/cmTC_42dc8.dir/src.c.o   -c /home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-Wshorten-64-to-32'
CMakeFiles/cmTC_42dc8.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_42dc8.dir/src.c.o' failed
make[2]: *** [CMakeFiles/cmTC_42dc8.dir/src.c.o] Error 1
make[2]: Leaving directory '/home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_42dc8/fast' failed
make[1]: *** [cmTC_42dc8/fast] Error 2
make[1]: Leaving directory '/home/user/Downloads/stlink-master/build/Release/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }

Any help solving this error is greatly appreciated.

xor-gate, 2017-02-01:
Your log is way to detailed but this one is you are looking for:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
```

Probably you have not installed the libusb-1.0-0-dev package (not sure the exact name). CMake could not find the libusb header files.

bmetman, 2017-02-01:
Oh wow, didn't even notice that. Thanks a lot, works perfectly now!

closed 2017-02-01 by bmetman

xor-gate, 2017-02-01:
No problem, we are all humans :+)

yota-code, 2017-04-05:
Yes ! thank you
Since it took me some iterations to solve this, for Ubuntu yaketty, here is the copy-paste procedure:
```
sudo apt install cmake libusb-1.0-0-dev
export LIBUSB_INCLUDE_DIR=/usr/include/libusb-1.0
make clean
make
```

xor-gate, 2017-04-05:
Is it necessary to export the LIBUSB_INCLUDE_DIR? CMake module which I wrote to find libusb should find it by itself.