stlink-org/stlink #554: .deb binary package for v1.3.x release (ubuntu 16 and debian 8)

xor-gate, 2017-02-07:
Probably many people would benefit from a debian package.
@nekromant could you maybe help out (probably a static linked package is the easiest)?

nekromant, 2017-02-07:
@xor-gate No problem, I'll see what I can do. As for a statically linked, I think it's better to provide that one as a .tgz and let all the debs be dynamically linked.

xor-gate, 2017-02-07:
Maybe you are right, but we need to be sure the user of the package must install all dependencies first before installing. But that speaks for itself.

Thanks!

xor-gate, 2017-02-22:
Hi @nekromant I have tried myself, but the libgtk+3 dependency is hardcoded. I'm not using this at all and the debianize seems to fail without it installed running:

```
debuild --no-lintian -us -uc -b -d
```

```
make[1]: Leaving directory '/var/lib/jenkins/workspace/Tools_stlink/obj-x86_64-linux-gnu'
   dh_install -O--buildsystem=cmake
dh_install: stlink-gui missing files (/usr/bin/stlink-gui*), aborting
```

and

```
dpkg-buildpackage: host architecture amd64
dpkg-checkbuilddeps: Unmet build dependencies: libgtk-3-dev
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -D -us -uc -b failed
```

nekromant, 2017-02-23:
As far as I remember it is listed as build-dependency, since when building the packages we (normally) want to build all components. As for the end-user the gtk dependency is only listed in the deb containing the ui, so unless he/she wants the gui, the package can be safely omited. 

Here, I quickly packaged the from 1.3.0 branch on my machine. It still thinks it's the 1.2.1 version though, I'll try to fix it by monday. 
For now - here go the debs. 
https://cloud.ncrmnt.org/index.php/s/2g2NVbYgpLmXxOw

xor-gate, 2017-02-28:
Thanks, I'm not 100% happy how debian packages are build. With CPack this is much easier and I have added initial integration: https://github.com/texane/stlink/commit/5b69f25198a1a8f34e2ee48d1ad20f79447e3d55

closed 2017-03-01 by xor-gate