stlink-org/stlink #555: Cannot connect with STM32F469I-DISCO board

soeugene, 2017-02-08:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [Stlink/v2-1 onboard] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [OSX Yosemite 10.10.5] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ 1.1.0] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ st-util] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ STM32F469I] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:
2017-02-08T11:33:26 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2017-02-08T11:33:26 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2017-02-08T11:33:26 DEBUG src/stlink-common.c: *** looking up stlink version
2017-02-08T11:33:26 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2017-02-08T11:33:26 DEBUG src/stlink-common.c: stlink pid     = 0x374b
2017-02-08T11:33:26 DEBUG src/stlink-common.c: stlink version = 0x2
2017-02-08T11:33:26 DEBUG src/stlink-common.c: jtag version   = 0x19
2017-02-08T11:33:26 DEBUG src/stlink-common.c: swim version   = 0xd
2017-02-08T11:33:26 INFO src/stlink-common.c: Loading device parameters....
2017-02-08T11:33:26 DEBUG src/stlink-common.c: *** stlink_core_id ***
2017-02-08T11:33:26 DEBUG src/stlink-common.c: core_id = 0x2ba01477
2017-02-08T11:33:26 DEBUG src/stlink-common.c: *** stlink_read_debug32 10006434 is 0xe0042000
2017-02-08T11:33:26 WARN src/stlink-common.c: unknown chip id! 0x10006434
2017-02-08T11:33:26 DEBUG src/stlink-common.c: *** stlink_close ***

```
OUTPUT/ERROR of the commandline tool(s)
unknown chip id! 0x10006434
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2017-02-08:
You are using a very very old version (1.1.0). Please use the latest released version (1.3.0) or the latest in the master branch.

xor-gate, 2017-02-14:
Closing due to inactivity. Consider trying a recent version. Feel free to open a new issue if the problem still remains. Thanks!

Nightwalker-87, 2020-03-19:
Please verify with Release v1.3.0.

closed 2020-04-06 by Nightwalker-87