stlink-org/stlink #556: ST-Flash fails randomly on Raspberry Pi 2

Galaxy88, 2017-02-10:
- [ X] Programmer/board type: Stlink/v2,
- [ X] Programmer firmware version: V2J25S4
- [ X] Operating system: Raspbian 8.0 Linux raspberrypi 4.4.38-v7+ 
- [ X] Stlink tools version and/or git commit hash: vv1.3.0-9-g88c6162
- [ X] Stlink commandline tool name: `st-flash`
- [X ] Target chip (and optional board): STM32F072CBU

I have a bug which appears on Raspbian (For info, it works perfectly on Windows 10 and Ubuntu 16.04). The st-flash will sometimes fail to program my target chip with either of  the error messages: 
home/pi/stlink/src/flash_loader.c: flash loader run error
ERROR /home/pi/stlink/src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1

OR:

INFO /home/pi/stlink.git/src/common.c: Loading device parameters....
WARN /home/pi/stlink.git/src/common.c: unknown chip id! 0xa05f0000

I have tried erasing the Flash before programming as detailed in the README but this does not prevent the problem, only makes it less prevalent. Have seen the other closed issues but none of them seem to present both messages at the same time. 

Thank you for your awesome tool.

Output:

```
INFO /home/pi/stlink.git/src/common.c: Loading device parameters....
WARN /home/pi/stlink.git/src/common.c: unknown chip id! 0xa05f0000
```

```
INFO /home/pi/stlink/src/common.c: Loading device parameters....
INFO /home/pi/stlink/src/common.c: Device connected is: F07x device, id 0x20016448
INFO /home/pi/stlink/src/common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 2048 bytes
INFO /home/pi/stlink/src/common.c: Attempting to write 42512 (0xa610) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800a000 erased
INFO /home/pi/stlink/src/common.c: Finished erasing 21 pages of 2048 (0x800) bytes
INFO /home/pi/stlink/src/common.c: Starting Flash write for VL/F0/F3 core id
INFO /home/pi/stlink/src/flash_loader.c: Successfully loaded flash loader in sram
ERROR /home/pi/stlink/src/flash_loader.c: flash loader run error
ERROR /home/pi/stlink/src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

xor-gate, 2017-02-10:
I must say, this is a very weird situation. I have seen people having trouble with a half broken USB cable. Not connected reset to target. You should try to upgrade the firmware of your programmer to V2J27 I have seen better results with this.

Galaxy88, 2017-02-11:
Hi thanks for the suggestion. Have upgraded the firmware of the programmer to V2J27 but the problem still persists randomly.

xor-gate, 2017-02-11:
To bad this didn't help. The difficult part is it happens "random", hopefully other people will spot this and can reproduce. Currently I have no idea or fix.

Nightwalker-87, 2020-02-23:
@Galaxy88 Could you try a different USB cable to rule out the point mentioned by @xor-gate ?
Please also test this with Release v1.4.0, v1.5.0, v1.5.1 and v.1.6.0, which have been released in the meanwhile, as this issue is already very old.

Nightwalker-87, 2020-03-24:
It would be helpful if somebody with a Raspberry Pi 2 could do some tests with old versions here to help find a related fix, should it exist already... Any volunteers?

Nightwalker-87, 2020-07-05:
Is this reproducible with a STM32F072 or any similar MCU in a comparable environment?
Tests should occur on the `v1.3.1-patch` branch at first.

chenguokai, 2020-07-09:
Since the flashing fails randomly and occurs on a RPI2, where undervolt occurs from time to time when it is connected with peripherals or running at a heavy load, I would suggest check the volt of the ST MCU.

Nightwalker-87, 2021-03-12:
Apart from the advice from @chenguokai this is also related to #356.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87