stlink-org/stlink #564: STM32L011G4 flash error (unknown coreid, write_half_pages failed)

zakqwy, 2017-03-02:
- [X] Programmer/board type: Stlink/v2-onboard (STM32F030 Discovery Board)
- [X] Operating system: Linux Mint 17.2
- [X] stlink version: v1.3.1/git-6f24bde
- [X] stlink commandline tool name: st-flash
- [X] Target chip: STM32L011G4

I am attempting to flash a simple 'hello-world' type LED blink program onto a bodged together STM32L011G4-based board, using my STM32F030 Discovery Board's on-board Stlink/v2. 

Command: `$ make helloworld.stlink-flash`

Output:

`  FLASH  helloworld.bin`
`st-flash 1.3.1`
`2017-03-02T16:27:46 INFO src/common.c: Loading device parameters....`
`2017-03-02T16:27:46 INFO src/common.c: Device connected is: L011 device, id 0x10086457`
`2017-03-02T16:27:46 INFO src/common.c: SRAM size: 0x800 bytes (2 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 128 bytes`
`2017-03-02T16:27:46 INFO src/common.c: Attempting to write 480 (0x1e0) bytes to stm32 address: 134217728 (0x8000000)`
`Flash page at addr: 0x08000180 erased`
`2017-03-02T16:27:46 INFO src/common.c: Finished erasing 4 pages of 128 (0x80) bytes`
`2017-03-02T16:27:46 INFO src/common.c: Starting Half page flash write for STM32L core id`
`2017-03-02T16:27:46 ERROR src/flash_loader.c: unknown coreid, not sure what flash loader to use, aborting! coreid: bc11477, chipid: 457`
`2017-03-02T16:27:46 WARN src/flash_loader.c: Failed to write flash loader to sram!`
`2017-03-02T16:27:46 WARN src/common.c: stlink_flash_loader_init() == -1`
`2017-03-02T16:27:46 WARN src/common.c: `
`write_half_pages failed == -1`
`  2/  3 pages written`
`2017-03-02T16:27:46 INFO src/common.c: Starting verification of write complete`
`2017-03-02T16:27:46 ERROR src/common.c: Verification of flash failed at offset: 0`
`stlink_fwrite_flash() == -1`
`make: *** [helloworld.stlink-flash] Error 255`

I got the same error using `st-util` and uploading the file manually. One thing I did notice -- when I first attempted to flash the chip the SRAM size was set to 0x4000 (i.e. 16 KiB); this was incorrect for the STM32L0 I am using so I changed line 455 of `chipid.c` to `.sram_size = 0x800,` and rebuilt the program. The flash memory size is correct for the G4 version.

I tried running `st-flash erase` and it seemed to work fine -- all 128 pages erased. I also tried shorting the BOOT0 pin (pin 1) to VCC and resetting the chip a few times, as suggested by other users experiencing similar-looking problems. This didn't seem to have any effect on subsequent programming attempts.

Perusing `common.c` I did notice the following lines (ln 1835-1836):
`else if (sl->flash_type == STLINK_FLASH_TYPE_L0) {`
`    /* use fast word write. todo: half page. */`

Does this mean that the half page write mode isn't implemented and I just need to switch over to programming by word? I looked around for an option to do this but I couldn't find one.

zakqwy, 2017-03-03:
Update: I realized that the commands I used above are specific to the libopencm3 examples. Here is the somewhat lengthy output of `st-flash` attempting to directly program a binary file with debug enabled:

`$ st-flash --debug write helloworld.bin 0x8000000`

st-flash 1.3.1
2017-03-03T09:53:13 DEBUG src/common.c: stlink current mode: mass
2017-03-03T09:53:13 DEBUG src/common.c: stlink current mode: mass
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_enter_swd_mode ***
2017-03-03T09:53:13 DEBUG src/common.c: *** looking up stlink version
2017-03-03T09:53:13 DEBUG src/common.c: st vid         = 0x0483 (expect 0x0483)
2017-03-03T09:53:13 DEBUG src/common.c: stlink pid     = 0x3748
2017-03-03T09:53:13 DEBUG src/common.c: stlink version = 0x2
2017-03-03T09:53:13 DEBUG src/common.c: jtag version   = 0xe
2017-03-03T09:53:13 DEBUG src/common.c: swim version   = 0x0
2017-03-03T09:53:13 DEBUG src/common.c:     notice: the firmware doesn't support a swim interface
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_jtag_reset ***
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_reset ***
2017-03-03T09:53:13 INFO src/common.c: Loading device parameters....
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_core_id ***
2017-03-03T09:53:13 DEBUG src/common.c: core_id = 0x0bc11477
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0xe0042000
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 10086457 is 0x40015800
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 3a80010 is 0x1ff8007c
2017-03-03T09:53:13 INFO src/common.c: Device connected is: L011 device, id 0x10086457
2017-03-03T09:53:13 INFO src/common.c: SRAM size: 0x800 bytes (2 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 128 bytes
2017-03-03T09:53:13 DEBUG src/common.c: *** set_swdclk ***
2017-03-03T09:53:13 DEBUG src/common.c: stlink current mode: debug (jtag or swd)
2017-03-03T09:53:13 DEBUG src/common.c: stlink current mode: debug (jtag or swd)
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_force_debug_mode ***
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_status ***
2017-03-03T09:53:13 DEBUG src/common.c:   core status: halted
2017-03-03T09:53:13 INFO src/common.c: Attempting to write 480 (0x1e0) bytes to stm32 address: 134217728 (0x8000000)
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_core_id ***
data_len = 2 0x2
 77 14

2017-03-03T09:53:13 DEBUG src/common.c: core_id = 0x0bc11477
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 208 to 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 0 to 0x8000000
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c18
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40023c04
Flash page at addr: 0x08000000 erased2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 208 to 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 0 to 0x8000080
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c18
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40023c04
Flash page at addr: 0x08000080 erased2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 208 to 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 0 to 0x8000100
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c18
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40023c04
Flash page at addr: 0x08000100 erased2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 208 to 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 0 to 0x8000180
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c18
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40023c04
Flash page at addr: 0x08000180 erased
2017-03-03T09:53:13 INFO src/common.c: Finished erasing 4 pages of 128 (0x80) bytes
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 89abcdef to 0x40023c0c
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 2030405 to 0x40023c0c
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 8c9daebf to 0x40023c10
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 13141516 to 0x40023c10
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 INFO src/common.c: Starting Half page flash write for STM32L core id
[... 244 lines cut ...]
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c18

2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40023c04
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 7 to 0x40023c04
[... cut ...]
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x8000004
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_exit_debug_mode ***
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2017-03-03T09:53:13 DEBUG src/common.c: *** stlink_close ***

xor-gate, 2017-03-03:
I have added the STM32L011 but not 100% tested (only tested probe). Probably the configuration is broken (needs to verified against datasheet which I did not yet):
https://github.com/texane/stlink/blob/master/src/chipid.c#L448-L458

zakqwy, 2017-03-03:
Thanks @xor-gate. I double-checked the `chipid` values against the datasheet and made two more changes per table 3 from RM0377:

`.bootrom_base = 0x1ff00000`
`.bootrom_size = 0x2000` 

However after rebuilding I still got the same error. I'll keep looking into the issue.

xor-gate, 2017-03-03:
Ping @gluedig, @mcoquelin-stm32 could you verify this (if you have a target?).

xor-gate, 2017-03-03:
Probably @tomdeboer has also some thoughts as he had some commit history on the L0 flash loaders.

zakqwy, 2017-03-15:
Okay, I think I've got a fix for this. In common.h, there are a number of `if` statements that check `chip_id` and set `flash_regs_base`. The statements default to `STM32L_FLASH_REGS_ADDR`if the `chip_id` doesn't match the query, which is the wrong address for the STM32L011 series (it should be `STM32L0_FLASH_REGS_ADDR`). I updated the appropriate lines to:
`if (sl->chip_id == STLINK_CHIPID_STM32_L0 || sl->chip_id == STLINK_CHIPID_STM32_L0_CAT5 || sl->chip_id == STLINK_CHIPID_STM32_L0_CAT2 || sl->chip_id == STLINK_CHIPID_STM32_L011)`. Locations in `common.c`: [here](https://github.com/texane/stlink/blob/master/src/common.c#L1470-L1474), [here](https://github.com/texane/stlink/blob/master/src/common.c#L1666-L1670), [here](https://github.com/texane/stlink/blob/master/src/common.c#L1841-L1847).

xor-gate, 2017-03-15:
@zakqwy you may propose a pull request and get free credits :+)

xor-gate, 2017-03-15:
Thanks, feel free to open a new issue or PR when something is not working as expected. Thank you for your contribution!

xor-gate, 2017-03-15:
See PR #572

closed 2020-03-18 by Nightwalker-87