stlink-org/stlink #568: sudden unknown chip id

tcurdt, 2017-03-13:
- [x] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [x] Operating system: macOS 10.12.3
- [x] Stlink tools version: v1.3.1
- [x] Stlink commandline tool name: `st-flash` and `st-util`
- [x] Target chip: generic STM32F103C8

I was flashing without problems, suddenly I am getting "unknown chip id".

Output:

```
$ st-flash write main.bin 0x8000000
st-flash 1.3.1
2017-03-13T01:22:48 INFO src/common.c: Loading device parameters....
2017-03-13T01:22:48 WARN src/common.c: unknown chip id! 0xe0042000
```

Did I somehow fry my board? Or what does that mean?

tcurdt, 2017-03-13:
This happens on an erase:
```
$ st-flash erase
st-flash 1.3.1
2017-03-13T01:49:32 INFO src/common.c: Loading device parameters....
2017-03-13T01:49:32 WARN src/common.c: unknown chip id! 0xe0042000
```
When I hold the reset button while calling the erase I get:
```
$ st-flash erase
st-flash 1.3.1
2017-03-13T01:49:40 INFO src/common.c: Loading device parameters....
2017-03-13T01:49:40 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2017-03-13T01:49:40 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0 bytes (0 KiB) in pages of 1024 bytes
Mass erasing
```

0 bytes?

tcurdt, 2017-03-13:
Setting boot0 to 1, then `st-flash erase`, then setting it back to 0 and all was working again.
I am fine to close the issue - but still would love to hear how/why that happened.

iglooom, 2017-04-09:
@tcurdt are you using SWDIO/SDCLK as GPIO in your firmware? It may be like #577

tcurdt, 2017-04-09:
@iglooom not by choice at least. I would need to check if any of the GPIO pins I am using are somehow connected to the SWDIO/SDCLK but according to this http://wiki.stm32duino.com/index.php?title=Blue_Pill they are not.

Nightwalker-87, 2020-03-20:
Reopening this for further investigation.

Nightwalker-87, 2020-05-27:
@tcurdt: I know this has been long ago, but du you still have the possibility to test if this is fixed now?

tcurdt, 2020-05-27:
I might take a while - but still have a blue pill I can check with.

Nightwalker-87, 2020-05-27:
If you do some testing, ensure you use the `develop` branch or Release v1.6.1 (once released) for testing.

Nightwalker-87, 2020-06-03:
Now as Release v1.6.1 is published, it's about time to do some testing here. We should go back in time following recent releases to find out if this still appears somewhere...

dkorkmazturk, 2020-06-11:
I have this exact problem with stm32f303k8 nucleo board with stlink 1.6.0 on Arch Linux. Programmer has firmware V2J37M26 (STSW-LINK007 3.3.4). It was working before but suddenly, I started to get the following error:

> st-util 1.6.0
> 2020-06-10T23:53:41 INFO common.c: Loading device parameters....
> 2020-06-10T23:53:41 WARN common.c: unknown chip id! 0x5fa0004

Since setting boot pins seems like requires some soldering on this board, I didn't get chance to test the solution suggested by @tcurdt.

I also tried it on Ubuntu 20.04 with stlink 1.6.1 and I get a different error

> st-util
> 2020-06-11T00:15:09 WARN common.c: Invalid flash type, please check device declaration
> 2020-06-11T00:15:09 ERROR gdb-server.c: Unsupported Target (Chip ID is 0000000000, Core ID is 0x2ba01477).

tcurdt, 2020-06-17:
I guess then I don't even need to test on my blue pill then. Seems like it's not fixed then.

Nightwalker-87, 2020-06-17:
Well a test with v1.6.1 would be useful indeed before adding it to the v1.6.2 milestone.

Nightwalker-87, 2020-06-17:
Well, a test with v1.6.1 would be indeed be useful before adding this ticket to the v1.6.2 milestone.

tcurdt, 2020-06-17:
Ah, missed that the other report was with 1.6.0.

JihedChaibi, 2020-07-16:
Same problem here.. with STM32F401RE Nucleo board on Ubuntu 18.04,
It was working perfectly but suddenly I started to get this error :

`$ st-flash write output/$(PROJ_NAME).bin 0x08000000`
`st-flash 1.6.1-69-gbe0157f`
`2020-07-16T02:48:35 WARN common.c: unknown chip id! 0x374b`

Nightwalker-87, 2020-07-17:
@JihedChaibi: Note the different error code `0x374b` in your warning.
I would need to investigate what exact cause is linked to this ID.

SJChannel, 2020-09-24:
I'm having a similar problem on Mac OS Catalina using st-flash 1.6.1 from Homebrew on a new fresh-out-of-the-box Nucleo-F746ZG.  Here is the output from "st-flash reset":

```
st-flash 1.6.1
2020-09-24T13:59:47 WARN common.c: unknown chip id! 0x5fa0004
Failed to connect to target

```

An old st-flash 1.3.0 executable on the same Mac works OK:

```
2020-09-24T13:58:40 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Loading device parameters....
2020-09-24T13:58:40 INFO /Users/jerry/Downloads/stlink-master/src/common.c: Device connected is: F7 device, id 0x10016449
2020-09-24T13:58:40 INFO /Users/jerry/Downloads/stlink-master/src/common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
st-flash 1.3.0

```

And st-flash 1.5.1 on a Raspberry Pi 4 running Raspian also works OK:

```
2020-09-24T13:53:02 INFO common.c: Loading device parameters....
2020-09-24T13:53:02 INFO common.c: Device connected is: F7 device, id 0x10016449
2020-09-24T13:53:02 INFO common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
st-flash 1.5.1

```

Nightwalker-87, 2021-03-12:
Time for a short summary:

We were originally talking about the following error: `WARN src/common.c: unknown chip id! 0xe0042000` which as we know by now definitely still appeared in v1.4.0 (see issue #669).

@dkorkmazturk @SJChannel this is a different error: `WARN common.c: unknown chip id! 0x5fa0004` the related error message is `Failed to connect to target` --> see issues #715, #791 and #1040 instead
@JihedChaibi  your error: `WARN common.c: unknown chip id! 0x374b` is not related to this issue either

In order to get ahead with this, I'll try to reproduce this issue for v1.4.0 using my Bluepill board and hope some people are still following this ticket and are able to give some useful feedback.

Nightwalker-87, 2021-03-12:
Hardware:
- Programmer STLink/V2-clone / Firmware V2.J37.S7
- Board: STM32F103C8T6 ("Bluepill")
- OS: Debian (Bullseye [Testing])

Due to an existing compilation error I could not use v1.4.0, but v1.3.1 (with a patch regarding version numbering, to be able to compile) instead. I did a read-write cycle with and without board resets in-between and could not reproduce the described behaviour:

```
$ st-info --probe
Found 1 stlink programmers
 serial: 3f70050132124647524b4e00
openocd: "\x3f\x70\x05\x01\x32\x12\x46\x47\x52\x4b\x4e\x00"
  flash: 65536 (pagesize: 1024)
   sram: 20480
 chipid: 0x0410
  descr: F1 Medium-density device

$ st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.3.1-1-gbea4915-dirty
2021-03-12T14:33:15 INFO src/common.c: Loading device parameters....
2021-03-12T14:33:15 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T14:33:15 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2021-03-12T14:33:15 INFO src/common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000c00 erased
2021-03-12T14:33:15 INFO src/common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T14:33:15 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2021-03-12T14:33:15 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  3/3 pages written
2021-03-12T14:33:15 INFO src/common.c: Starting verification of write complete
2021-03-12T14:33:15 INFO src/common.c: Flash written and verified! jolly good!

$ st-flash erase
st-flash 1.3.1-1-gbea4915-dirty
2021-03-12T14:33:45 INFO src/common.c: Loading device parameters....
2021-03-12T14:33:45 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T14:33:45 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
Mass erasing

[ Cycle repeated several times (with and without manual board resets in-between)... ]

$ st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.3.1-1-gbea4915-dirty
2021-03-12T14:33:56 INFO src/common.c: Loading device parameters....
2021-03-12T14:33:56 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T14:33:56 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2021-03-12T14:33:56 INFO src/common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000c00 erased
2021-03-12T14:33:56 INFO src/common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T14:33:56 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2021-03-12T14:33:56 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  3/3 pages written
2021-03-12T14:33:56 INFO src/common.c: Starting verification of write complete
2021-03-12T14:33:56 INFO src/common.c: Flash written and verified! jolly good!

$ st-flash erase
st-flash 1.3.1-1-gbea4915-dirty
2021-03-12T14:34:01 INFO src/common.c: Loading device parameters....
2021-03-12T14:34:01 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T14:34:01 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
Mass erasing

$ st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.3.1-1-gbea4915-dirty
2021-03-12T14:34:12 INFO src/common.c: Loading device parameters....
2021-03-12T14:34:12 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-12T14:34:12 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2021-03-12T14:34:12 INFO src/common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000c00 erased
2021-03-12T14:34:12 INFO src/common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T14:34:12 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2021-03-12T14:34:12 INFO src/flash_loader.c: Successfully loaded flash loader in sram
  3/3 pages written
2021-03-12T14:34:12 INFO src/common.c: Starting verification of write complete
2021-03-12T14:34:12 INFO src/common.c: Flash written and verified! jolly good!
```

Looking at this, the observed behaviour does not seem to be related to a bug present in the toolset in that version.

Nightwalker-87, 2021-03-13:
ToDo: https://github.com/stlink-org/stlink/issues/669#issuecomment-660553861

Nightwalker-87, 2021-03-20:
Related to #443.

Nightwalker-87, 2021-03-20:
A solution to this can be found here: https://github.com/stlink-org/stlink/issues/107#issuecomment-60971296. I'll add these steps to our tutorial.

@Ant-ON I'm still trying to figure out the actual technical meaning of `0xe0042000`.

This is what I found out so far:
- chip id: 0xE0042000 or 0x40015800, primary information to derive flash and SRAM architecture
- Chip IDs are explained in the appropriate programming manual for the DBGMCU_IDCODE register (0xE0042000)

So I assume this is simply the returned memory address where the chip ID would have been found, if a read-out was possible.
Does this imply that this address is always returned, if this register could not be read and thus it was not possible to display a valid chip ID like `0x430` (`STLINK_CHIPID_STM32_F1_XL`) etc. ?

For illustration:
- **OK**: `st-info --probe` --> Send request to read `0xE0042000` --> Try to read memory address `0xE0042000` --> Successfully read value in `0xE0042000` --> Send chip ID value back to console --> Display value
- **FAIL**: `st-info --probe` --> Send request to read 0xE0042000 --> Try to read memory address `0xE0042000` --> Failed to connect and thus access `0xE0042000` --> Return register address back to console --> Display memory address

Ant-ON, 2021-03-22:
@Nightwalker-87 chip id can be found in the `reference manual` as the `DBGMCU_IDCODE`. It's uniqual id of one or group of chips. st-link dterminea the support of MCU by chip id. `st-info` don't display information about a chip if read incorrect value from the `DBGMCU_IDCODE` register. Also posible `st-info` reads a value from incorrect register:
https://github.com/stlink-org/stlink/blob/40e9aa22bab024ef1ef4416dc854967504d8bf58/src/common.c#L1221-L1249
We first try read id from `0xE0042000`. We may got garbage if chip id contains in the `0x40015800` register. A more precise definition can be obtained using a code like https://github.com/stlink-org/stlink/pull/1101#issuecomment-802558712

Nightwalker-87, 2021-03-22:
As I understand the reference, the improvement is that `st-info` allocates and checks the correct register depending on the `cpu_id`, which is indeed an improvement. So in reference to your last comment this issue should then actually be verified with PR 1101.

commit 229c7211 (2021-03-24): General Project Update

- Added basic security policy
- Updated info on version support for macOS
- Updated tutorial.md
> Added reference for st-info --probe command
> Added section on unknown chip id error
   (Closes #107, Closes #568, Closes #669)
> Added information on HW reset pin issue
   (Closes  #220, Closes #238)
closed 2021-03-24 by Nightwalker-87